svvitale/nxppy #5: Authenticate method?

arthurv, 2015-02-08:
Are there any plans to include an authenticate() method in addition to the read() and write() methods?

This is for Mifare Classic cards...thanks!

svvitale, 2015-02-09:
Hi @arthurv!  I've had a couple of other requests for Mifare Classic support, but haven't really allocated any time to work on it.  There are some reasonable examples from NXP, so I don't think it's that large of an effort.  If you were interested in doing the work, I'd be happy to include the pull request!  Short of that, you're at the mercy of my free time.  I'll start by trying to get my hands on a Mifare Classic card (all I've used for my projects are ultralights).

arthurv, 2015-02-10:
It also depends on my free time but I'll have a go at it. I'm downloading the card_polling code now, and one thing I noticed were the separate commands for read() and write(): mifare_ultralight and mifare_classic, so maybe the python interface could take an argument for read() or write() depending on the type of card. I'll see what I can do.

svvitale, 2015-02-15:
The example code I found looks fairly straightforward: http://pastebin.com/UznGKDbC

I think I'd prefer to make a new python class "MifareClassic" that would support the authentication methods, separate from the existing Mifare code.  Of course, there's some overlap, but having it functionally separate from Python's perspective is a nice abstraction.

Another poster was kind enough to offer me some Mifare Classic cards (by mail), so hopefully I'll have something to play with soon.  Look forward to collaborating with you on this!

arthurv, 2015-02-15:
Sounds good!

So my intended strategy is to copy the existing python class, call it "MifareClassic", swap phalMful_Write() with phalMfc_Write() and adjust the required arguments, and do the same for the read() methods and create a new one for the authenticate() method using phalMfc_Authenticate.

For NDEF data it could be a wrapper on top of those methods to set the MAD sectors, and write the data eg. URIs in the format, but for now I'm only interested in byte-level reading and writing and having 1K of memory instead of just 64 bytes :)

svvitale, 2015-02-15:
Sounds great!  If there are common functions (I'm guessing mifare_init will be identical), feel free to move them into a MifareCommon.h/c.  Really appreciate the work, Arthur, you're not the first that's asked for this!

svvitale, 2015-02-24:
Not sure how far you've gotten, @arthurv, but wanted to let you know that there are some larger changes coming.  The current method I'm using is not supported in the latest Raspian release, so I'm working with NXP to make some modifications.  I'd be more specific, but I don't know the specifics yet ;).

arthurv, 2015-02-27:
Hi @svvitale , things have been quite busy on my end so not much progress besides laying out the strategy :) Look forward to hearing about the modifications!

svvitale, 2015-03-04:
Updates are all in.  Not much has changed in the code itself, most of the changes were in the build process.  I have some Mifare Classic tags on their way to me now (thanks to the good folks at NXP).  I should be able to give you a hand with testing once they arrive.

toby314, 2015-03-04:
thanks guys for your work. I am trying to use nxppy for micro-credit transactions on MIfare 1k.  I have no idea where to start with securing (Hashing) the user data (credit) on the cards. is it built into NFC protocols ?  great work SV ! thank you.

svvitale, 2015-03-05:
Hi @toby314, unfortunately I don't know anything about how to secure micro-credit transactions over NFC.  I imagine that using the encryption mechanisms for Mifare Classic will help, but I don't know enough to safe if it will be fully secure.  Most of the secure payment systems are using EMV, which I know even less about.  Sorry I can't be of more help.

djambo, 2015-08-27:
Hi @svvitale, did you had the chance to work with those Mifare Classic yet? I am trying so hard to make it work with these little guys for an internal project we are doing in my office, but I have not much knowledge of c.

techpet, 2015-08-27:
Hello @svvitale I would also like to know if there was any progress with Mifare Classic authentication method in python. I'm rather curious why there doesn't yet exist a complete python library to write to Mifare Classic tags despite of NXP-EXPORE board support for them. Thank you!

birogeri, 2015-09-20:
Hi! Nxppy is awesome! I would like to ask about the Mifare Classic authentication method, too. I hope you can manage to set aside some free time for it. Thanks!

AlterCodex, 2015-09-29:
Hi there, Nxppy its an awesome project, right now I need to implement all the functions for mifare classic 1k and 4k and I'm active working on that using @techpet strategy when I finish it  would be nice to give the sources to anyone who needs it :)

birogeri, 2015-09-30:
That would be awesome!

AlterCodex, 2015-10-14:
right now I'm working on a stable version but already have the authenticate method working next is the read and write, I must speak with @svvitale  because I'm using a of of his codes https://github.com/AlterCodex/Nxppy2/tree/Develop

svvitale, 2015-10-14:
Hi @AlterCodex!  Thanks for your work on this.  Would you mind forking nxppy and then filing a pull request with your changes?  I'd like to integrate them back into the mainline after we're both satisfied with them.

AlterCodex, 2015-10-14:
Hi @svvitale no problem I will do it after it's finished ;) (maybe I will need some help doing it I'm new in git normally uses tfs)

svvitale, 2015-10-14:
Github makes it easy.  Just go to my nxppy repository and click the "fork"
button.  Now you've got a copy of the repository under your own name that
you can make commits to.  Create a pull request to offer your commits back
to me :).

AlterCodex, 2015-10-20:
It's done i hope it help someone :)

birogeri, 2015-10-25:
I built and tested it on my Rpi, compiles & works OK on my Mifare 1Ks! Thank you!

birogeri, 2015-10-26:
@AlterCodex Two questions:
* How can I erase (eg. write 0s to a part of a block)?
 The following code drops `TypeError: must be string without null bytes, not str`
```python
#I want this to be on the card: 1111 1111 0000 0000 1111 1111
  data = "FF00FF".decode("hex")
  result = mifare.classic_write(x,data)
```
* Writing less than 8 bytes results in strange values written to the block besides the intended data
```python
#I want this to be on the card: 54 65 73 74 54 65 00 00 00 ...
  data = "54657374".decode("hex")
# Or (0x54 0x65 0x73 0x74 should be the same as --> "Test")
  data = "Test"
  result = mifare.classic_write(x,data)
```
But the block is going to be `5465737400616C7565000000...` in the first case (Test[0x00]alue[0x00]...) and `54657374000000000000000004000000` in the second case. Do you have any ideas why?

AlterCodex, 2015-10-26:
@birogeri, I dont know why you get that error tomorrow I will check it, for the second one remember that nxppy, specially the mifare classic part its just a wrapper for the c++ code, searching on the documentation I get that :
WriteValue - phalMfc_WriteValue() 
Firstly, this function creates from 4 byte input value and input address the 16 byte formatted structure according to Fig 4. Then it performs MIFARE Classic Write command of that 16 byte value. 
Formatting is done automatically by MCU software - phalMfc_Int_CreateValueBlockFormat(). phStatus_t phalMfc_WriteValue(      void * pDataParams,         [In]      uint8_t bBlockNo,         [In]      uint8_t * pValue,         [In]      uint8_t bAddrData )       
  [In] *pDataParams: pointer to the MIFARE Classic AL layer data parameter structure.
 bBlockNo: block number to be written into.
 *pValue: 4 byte array to be written to the MIFARE Classic card value block. This function converts and includes it to the 16 byte format ready to be written. 
bAddrData: one byte array of address data to be written to the MIFARE Classic card value block. This function converts and includes it to the 16 byte format ready to be written. 
It always writes 16 bytes, so if you send aniyhig less that 16 the Cpp code will take the memory bytes near your pointer to the array of data, I will add in the future a extra validation to prevent this, but for now send always 16 bytes to the write function.

birogeri, 2015-10-30:
@AlterCodex (NB I don't know how to program in C++). I looked around, and I think the problem is that you use [ParseTuple here](https://github.com/AlterCodex/nxppy/blob/MifareClassic/Mifare.c#L243) with 'by' format, while @svvitale uses 'by#' which creates [two variables](https://docs.python.org/3/c-api/arg.html?highlight=parsetuple#strings-and-buffers), one is the same as y _(but accepts NUL chars!)_, and one with its length, but the latter is not used for anything in Scott's part of the code.

svvitale, 2015-11-04:
@birogeri, I'll be integrating @AlterCodex's pull request soon.  I'll likely make a few consistency changes and I'll keep an eye on the "erase" use case specifically.

electricmill, 2016-06-05:
Thanks @svvitale and @AlterCodex for this great work. I installed @AlterCodex  work for the MifareClassic this morning - it complained about the absence of SW282816.zip so downloaded this and modified the shell script.

I am reading from block 1 of the 4k mifare - with the sameple code you provided - but it does seem to read differing data from the block; I've checked on an android app and only block 1 is populated. any idea where the rogue data might be  coming from?

Gadouille, 2017-02-02:
Hello, 
I reloaded nxppy to get Altercodex branch for 1K mifare, yet the get_type method is still not known. I wonder if sudo pip install nxppy is sufficient to get this branch?
Thanks for your work.
Regards

AlterCodex, 2017-02-02:
the latest  low level libs from nxp doesn't works with my branch, I will need to update my sources to make it compatible with them, but I'm really short on time for now if you can get and old zip file with the sources maybe you can use my branch

Gadouille, 2017-02-05:
Thanks for your answer,
I had neard-explorenfc_0.4-1_armhf.deb, I replaced it by neard-explorenfc_0.1-1_armhf.deb, but I still have the problem (_AttributeError: 'nxppy.Mifare' object has no attribute 'get_type'_). Can you tell me which version shall be working?
Otherwise, there is no hurry, If you intend to update the source one day, I can wait.
Thank you very much.

donikuy, 2017-02-28:
Hi,

I am trying to download and install nxppy for the EXPLORE-NFC kit but always encounter issues.

When i type "sudo pip install nxppy" the error i receive is "Cannot fetch index base URL https://pypi.python.org/simple/
Could not find any downloads that satisfy the requirement nxppy".

Also it does not work with git clone command.

Any ideas why i get this error?
[image]

AlterCodex, 2017-02-28:
When I was studying I had the same error because in my college were some proxy's and security protocols in the network that didn't make possible the connection to github.

donikuy, 2017-02-28:
Hi,

Thanks. I think that is what I am running into as well. Need to figure out what to type to get through that.
Or can I download the files and install manually?

svvitale, 2017-03-06:
@donikuy, please shoot me an email or file a separate issue if you're still having trouble.  This conversation is unrelated to issue #5.

serfati5, 2017-03-10:
Hi, I'm trying to read/write more than UID from Mifare Classic 4K. I don't get how to do it cause the segments are different than 1K cards.
Should I need this authenticate method? Did you get something more about it @svvitale @AlterCodex?

Here I have what I want to replicate more or less but it's using C# instead.
https://www.cooking-hacks.com/documentation/tutorials/rfid-13-56-mhz-nfc-module-arduino-raspberry-pi-tutorial/

Any help? I'm new in this. Thanks guys for your work!

svvitale, 2017-03-14:
Yes, @serfati5, all Mifare Classic cards will require this authentication before they're usable with nxppy.  Unfortunately I haven't had time to look at re-implementing this with the newer NXP reader library.

AlterCodex, 2017-03-30:
@svvitale @serfati5 @Gadouille sorry for the delays in this issue I just started to work on this again, in my branch is already the merge but it doesn't work :'( I have some spare time this week and I hope that by the end of next week the project will be running again.

serfati5, 2017-04-23:
Hi!! @svvitale @AlterCodex  thanks for your answers though.
Any news about it? Is it working already?

Thanks!!!

chlangbein, 2017-08-01:
Any news on this ? I've checked out https://github.com/AlterCodex/nxppy and could compile it if I change Mifare.c
345c345
<       uint8_t t= sDiscLoop.bDetectedTechs;
---
>       uint8_t t= self->nfcData.sDiscLoop.bDetectedTechs;
but it looks like the repository is not complete, I'm missing classic_read and classic_write methods. Any hints for this ?

AlterCodex, 2017-08-01:
Hi @chlangbein, some time ago I had to stop developing this because the new npx files (the zip that has all the libraries ) doesn't work with our connection board :( and we have like 200 hundreds of rpi and explore boards running as a product, so we have to keep  them running in  a really old source versions. But if you want to finish the develop maybe I can help you, my email is: fjb.pacho@gmail.com