svvitale/nxppy #6: nxppy.SelectError: FieldReset command failed: 20b

Korzonco, 2015-02-16:
Hi svvitale,

Thanks for your great work. I have been using your wrapper since last year.
It worked great so far but last week. I tried to start up a new project. 
I tried to go through a simple test, to get myself familiar with it but I gave me an error. 
This is my testing script, test.py, as follow:

import nxppy

mifare = nxppy.Mifare()
uid=mifare.select()
print uid

it shows this error:
Traceback (most recent call last):
  File "test.py", line 4, in <module>
    uid = mifare.select()
nxppy.SelectError: FieldReset command failed: 20b

Could you mind get me some idea how to fix it?

jwellnitz, 2015-02-18:
Hi - I'm new to raspberry pi and the NFC board - so I certainly could be doing something wrong - but I'm having the exact same issue (same script too). Note I am using the new Raspberry Pi 2 B.

svvitale, 2015-02-18:
Hi @Korzonco, @jwellnitz,

Note that the interface changed to be more pythonic in version 1.1 (including raising exceptions on error).  I just added an example to the README that shows how to do a simple polling loop.  Please let me know if that doesn't work for you.

https://github.com/svvitale/nxppy

Korzonco, 2015-02-20:
I tried following the README but same error appear even I use another mirco SD with a new image on it. Tried with both presenting a card or no card as well.

svvitale, 2015-02-20:
Hi @Korzonco,

Is this the same EXPLORE-NFC that you've been using all along?  Can you verify that SPI is enabled?

Korzonco, 2015-02-20:
Yes I did enable it. By sudo raspi-config.

>
> Hi @Korzonco,
>
> Is this the same EXPLORE-NFC that you've been using all along? Can you
verify that SPI is enabled?

jwellnitz, 2015-02-20:
I tried the new example too and it would never print the uid. I thought that was because I don't have anything written to the NFC tag. So I tried the same example, replacing mifare.select with mifare.write_block(10, b'abcd') and got the error WriteError: Write failed: 20b.

svvitale, 2015-02-21:
Well, I have good news and bad news.  The good news is that I reproduced this issue with Raspian 01-31-2015.  The bad news is that I don't have a fix yet.  For now, you can revert back to Raspian 12-24-2014 and things will start to work again.  The older version is available here: http://downloads.raspberrypi.org/raspbian/images/raspbian-2014-12-25/

I'll let you guys know as soon as I have a fix for newer versions.

Korzonco, 2015-02-23:
Hi svvitale,

Thanks for your support!!
I was thinking is it about the GPIO connection with rpi2. I had this issues in my other projects when rpi2 was released. And there is another possibility that NXP just releases a new source files. Hope to hear your good news. :)

jwellnitz, 2015-02-23:
I tried Scott's recommendation over the weekend (to go back to the older Raspian) and here are my results. Shorter version: it worked. 

- Downloaded older Raspian image
- Raspberry Pi 2 B wouldn't boot with it, I had a Raspberry Pi 1 A+ that would
- My USB hub that worked with the Pi 2 B wouldn't work with the PI 1 A+
- Went to store for new USB hub, it worked with the A+
- Tried to install python as recommended in the Readme, it failed (turns out I had to run apt-get update first, but I was afraid that would make the older Raspian image not work with nxppy again - see a few steps down)
- Installed cmake
- Tried to install nxppy - had to install pip first (found guidance online, something like: simple_update pythontools), not sure this would have been necessary if Python installed the first time)
- ran apt-get update, despite my fears, it didn't break anything
- tried again to install python, it worked
- tried again to install nxppy, it worked

I'm now a few steps farther along in my project. Thanks Scott.

It does seem like a raspian image update caused NXP to do some rearchitecting of their sw too, see:
http://www.element14.com/community/thread/41060/l/card-polling-problem

svvitale, 2015-02-24:
Thanks for verifying, @jwellnitz. I'm working with NXP directly on some of this, so after we get over this hump, I expect smooth sailing ahead.  I'll update this issue as I make progress.

toby314, 2015-03-01:
Hi
I'm loving nxppy (I'm using it with the older image 14.12.25 ). Soon I'm getting a raspberry pi 2 though. Ive found out that iw dosent woek with the older images, including 14.12.25. for a school project were making a water dispensing machine using a raspberry pi and NFC tags. Using the raspberry pi 2 for this would be great. Im wondering how long it might take for nxppy to be updated for use with newer images. if it could be working soon we will stick with the raspberry pi 2 for our project but if not we will have to use the raspberry pi B (we need the extra GPIO pins). Any insight into how close you are to making this work would be great. Thank you very much.

svvitale, 2015-03-01:
Hi @toby314,

The updates are complete on my test machine, but there are still some
changes needed in the build process.  I expect to release something this
week that will work with the Pi2.

-Scott

toby314, 2015-03-02:
Hi, its me again.
I've noticed that you can only write 4 bytes to blocks 10 and 14, i was hoping that i would be able to use more than that. Is this a problem with my pi or nxppy generally? I've also found that block 6 seems to work, is there a problem with doing I'm? i'm using this for a micro-payment application, can you shed any light on using nxppy for decreaing credit on the nyfare MIFARE card?
Thanks.

Korzonco, 2015-03-03:
Hi Toby,

I would suggest you to get a datasheet of your NFC tag. It should give you the information you need.

Cheers

michbil, 2015-03-03:
Having same problem on Raspberry PI 2, hope for fast fix.

svvitale, 2015-03-04:
This issue is now resolved in the 1.3.0 release.  Please refer to the README for some updated installation instructions, and let me know if you run into any issues.  Unfortunately I don't have a Pi2 to test on, but I have confirmed that nxppy is working again in the latest Raspian release (2015-02-16) on a Pi1 model B.

closed 2015-03-04 by svvitale

michbil, 2015-03-04:
Compiled from sources on Rpi2, it works. Thank you.

Korzonco, 2015-03-05:
Hi svvitale,

Thanks for your help.
My little rpi B+ with nxp & nxppy is back on live now.

Cheers :)