svvitale/nxppy #7: After 1.3.0 release NxpRdLib-Release is missing in repository

michbil, 2015-03-04:
When trying to install via pip i'm getting this 
In file included from Mifare.c:1:0:
    Mifare.h:31:25: fatal error: ph_NxpBuild.h: No such file or directory
    compilation terminated.
    error: command 'gcc' failed with exit status 1

svvitale, 2015-03-04:
Hi @michbil, do you have a full pip log?  I'm interested to see what happened prior to the compilation step.  It should include logging information about the package download and building of nxprdlib separately.

toby314, 2015-03-04:
this might help.

installed a raspberry Pi 2 (wheezy feb 2015) from scratch - here is the log file of commands entered:

same problem - help ????

>>>
lxsession-edit

sudo reboot
sudo dpkg -i libneardal0_0.14-1_armhf.deb libwiringpi2-2.25-1_armhf.deb neard-explorenfc_0.1-1_armhf.deb
wget https://bootstrap.pypa.io/get-pip.py
sudo python get-pip.py
sudo pip install nxppy
# Python.h not found
sudo apt-get update
sudo python get-pip.py
sudo pip install nxppy
wget https://bootstrap.pypa.io/get-pip.py
sudo apt-get install python-setuptools
cd ~/Downloads/nxppy-*
sudo python setup.py build install
sudo apt-get install cmake
sudo reboot
sudo pip install nxppy
#####
In file included from Mifare.c:1:0:
    
    Mifare.h:31:25: fatal error: ph_NxpBuild.h: No such file or directory
    
    compilation terminated.
    
    error: command 'gcc' failed with exit status 1
    
    ----------------------------------------
    Command "/usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-VcnAht/nxppy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-S4Z5gK-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-VcnAht/nxppy

toby314, 2015-03-04:
Forgot to post this earlier, hope this can help as well.

Collecting nxppy
  Using cached nxppy-1.3.0.tar.gz
    /usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py:2512: PEP440Warning: 'nose (dev)' is being parsed as a legacy, non PEP 440, version. You may find odd behavior and sort order. In particular it will be sorted as less than 0.0. It is recommend to migrate to PEP 440 compatible versions.
      PEP440Warning,
    Installed /tmp/pip-build-e0IgGI/nxppy/.eggs/nose-1.3.4-py2.7.egg
Installing collected packages: nxppy
  Running setup.py install for nxppy
    compiling NxpRdLib
    /bin/sh: 1: ./get_nxpRdLib.sh: not found
    building 'nxppy' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -Ineard-explorenfc-0.1/nxprdlib/types -Ineard-explorenfc-0.1/nxprdlib/intfs -I/usr/include/python2.7 -c Mifare.c -o build/temp.linux-armv7l-2.7/Mifare.o -O1
    In file included from Mifare.c:1:0:
    Mifare.h:31:25: fatal error: ph_NxpBuild.h: No such file or directory
    compilation terminated.
    error: command 'gcc' failed with exit status 1
    Complete output from command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-e0IgGI/nxppy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-sZsz_a-record/install-record.txt --single-version-externally-managed --compile:
    running install
    
    running build
    
    compiling NxpRdLib
    
    /bin/sh: 1: ./get_nxpRdLib.sh: not found
    
    running build_ext
    
    building 'nxppy' extension
    
    creating build
    
    creating build/temp.linux-armv7l-2.7
    
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -Ineard-explorenfc-0.1/nxprdlib/types -Ineard-explorenfc-0.1/nxprdlib/intfs -I/usr/include/python2.7 -c Mifare.c -o build/temp.linux-armv7l-2.7/Mifare.o -O1
    
    In file included from Mifare.c:1:0:
    
    Mifare.h:31:25: fatal error: ph_NxpBuild.h: No such file or directory
    
    compilation terminated.
    
    error: command 'gcc' failed with exit status 1

svvitale, 2015-03-04:
Fixed.  Unfortunately I had to bump the revision to 1.3.1 because pypi.  Let me know if pip doesn't work now.  Thanks!

closed 2015-03-04 by svvitale

toby314, 2015-03-04:
fixed thank you