svvitale/nxppy #10: Does nxppy supports NXP-EXPLORE card being used via UART?

jctlmm, 2015-03-11:
Hi,

I used a tutorial from this website https://learn.adafruit.com/downloads/pdf/adafruit-nfc-rfid-on-raspberry-pi.pdf to test my Adafruit NFC on Raspberry Pi and everything worked fine.

Now, I would like to use your extension from Python, but on your website there is a requirement that nxppy relies on SPI being enabled. Do you have any plan to extend nxppy so it can be used via UART as well?

Is this a task that I could do by myself if I modify your source code?

I really appreciate your inputs on this.

Thanks,

-Juan

svvitale, 2015-03-12:
Hi @jctlmm,

The nxppy module is specifically for the NXP EXPLORE-NFC board, which currently only works over SPI.  For the Adafruit NFC, I would recommend checking out either libnfc or nfcpy.  Here's a specific example of someone using the Adafruit board on a Raspberry Pi with nfcpy: https://answers.launchpad.net/nfcpy/+question/237349

Good luck!

closed 2015-03-12 by svvitale

jctlmm, 2015-03-12:
Thanks for the fast response and the feedback!!!