svvitale/nxppy #11: Installation problem on B+

lazy7, 2015-03-14:
After 
$wget https://bootstrap.pypa.io/get-pip.py
$sudo python get-pip.py

$sudo pip install nxppy
I am getting this ...
......
[Building NXP Reader Library]

    make: *** No targets specified and no makefile found.  Stop.

    running build_ext

    building 'nxppy' extension

    creating build

    creating build/temp.linux-armv6l-2.7

    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -Ineard-explorenfc-0.1/nxprdlib/types -Ineard-explorenfc-0.1/nxprdlib/intfs -I/usr/include/python2.7 -c Mifare.c -o build/temp.linux-armv6l-2.7/Mifare.o -O1

    In file included from Mifare.c:1:0:

    Mifare.h:23:20: fatal error: Python.h: No such file or directory

    compilation terminated.

    error: command 'gcc' failed with exit status 1

    ----------------------------------------
    Command "/usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-lUaGcz/nxppy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-dVljzV-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-lUaGcz/nxppy

I don't really know what is wrong.

svvitale, 2015-03-16:
Hi @lazy7,

Can you try installing the Python 2.7 development tools:

```sh
sudo apt-get install python2.7-dev
```

And then run the install again:

```sh
sudo pip install nxppy
```

Let me know if that doesn't work for you.

-Scott

lazy7, 2015-03-22:
Thank you! It worked like a charm.

closed 2015-03-22 by svvitale