svvitale/nxppy #12: Problem Installing, ph_NxpBuild.h not found

Xisku, 2015-03-18:
Hi,

Trying to install nxppy on Raspberry B this error occurs:

`pi@raspberrypi ~ $ sudo pip install nxppy`
`Collecting nxppy`
`  Using cached nxppy-1.3.1.tar.gz
    /usr/local/lib/python3.2/dist-packages/pkg_resources/__init__.py:2512: PEP440Warning: 'nose (dev)' is being parsed as a legacy, non PEP 440, version. You may find odd behavior and sort order. In particular it will be sorted as less than 0.0. It is recommend to migrate to PEP 440 compatible versions.
      PEP440Warning,
    Installed /tmp/pip-build-6vycsv/nxppy/.eggs/nose-1.3.4-py3.2.egg
Installing collected packages: nxppy
  Running setup.py install for nxppy
    compiling NxpRdLib`

{...Lots of successfull install lines...}

`running build_ext`
 `building 'nxppy' extension`
`    creating build`
`    creating build/temp.linux-armv6l-3.2`
`    gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -D_FORTIFY_SOURCE=2 -g -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -fPIC -Ineard-explorenfc-0.1/nxprdlib/types -Ineard-explorenfc-0.1/nxprdlib/intfs -I/usr/include/python3.2mu -c Mifare.c -o build/temp.linux-armv6l-3.2/Mifare.o -O1`

    In file included from Mifare.c:1:0:

    Mifare.h:31:25: fatal error: ph_NxpBuild.h: No such file or directory

    compilation terminated.

    error: command 'gcc' failed with exit status 1

    ----------------------------------------
    Command "/usr/bin/python3 -c "import setuptools, tokenize;__file__='/tmp/pip-build-6vycsv/nxppy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-ex0og0-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-6vycsv/nxppy`

I'm assuming that I need to install some software from NXP-Explore, but I dont know what.

Some help needed.

Thanks in advance!

svvitale, 2015-03-22:
Hi @Xisku,

Can you take a look in /tmp/pip-build-6vycsv/nxppy/ and see if there is a neard-explorenfc-0.1 directory?  The NXP-Explore libraries are all downloaded and extracted during the build, so there's no pre-requisite to install.  Sounds as if something during that process failed on your individual system.  From the same directory I referenced, you can run "sudo ./get_nxpRdLib.sh" to retry the whole download/extraction process.  Let me know what you find.

-Scott

Xisku, 2015-03-23:
Hi @svvitale,

Thanks for answering.

There is nothing left in /tmp after the install as it erases everything. Although, while installing, I tried to monitor which files were appearing and disappearing and neard-explorenfc-0.1 shortly appeared.
Also I'm not able to download/extract again for the same reason.

I remembered later that I was using python 3.2 and you say 2.7 or 3.4 should be used, so trying to go back to 2.7 I broke something. This afternoon I'll erase Raspbian and retry from the beggining with a clean install. Will tell you something then.

Thanks again!

svvitale, 2015-03-24:
Hi @Xisku,

Yes, you're right about /tmp, forgot about that.  You can use the "--no-clean" pip option to keep those directories if the error is reproducible.

I did verify that the existing nxppy package works with the installed python 3.2 provided you install the python3-dev package first:

```sh
sudo apt-get install python3-dev
```

Let me know if you still can't get things working.

-Scott

Xisku, 2015-03-25:
Hi @svvitale ,

I ran from a clean install of Wheezy Raspbian Feb 2015 and succesfully installed.
I only tested an import and single call to nppy.Mifare(), but seems OK!

Probably, as my Raspberry had been on the shelf for a time, it wasn't fully upgraded or something of a previous installation messed with it.

This time, after installing I just did the following:
-Enable SPI
``sudo apt-get install python2.7-Dev``
``sudo apt-get install python-setuptools``
``sudo apt-get install cmake``
``sudo apt-get install gcc``

And then your instructions:

``wget https://bootstrap.pypa.io/get-pip.py``
``sudo python get-pip.py``
``sudo pip install nxppy``

Will try something more at home tonight but It's done! Thank you!

svvitale, 2015-03-25:
Glad to hear it!

closed 2015-03-25 by svvitale