svvitale/nxppy #14: Can't install nxppy on Raspberry pi B+

ejeanneau, 2015-04-12:
Hi !

I can't install nxppy on my raspberry pi b+ (Raspbian), here the commands used : 

- ENABLE SPI
- apt-get updat
- sudo apt-get install python2.7-Dev
- sudo apt-get install python-setuptools
- sudo apt-get install cmake
- wget https://bootstrap.pypa.io/get-pip.py
- sudo python get-pip.py
- sudo pip install nxppy

==>
Collecting nxppy
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:79: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
  Using cached nxppy-1.3.1.tar.gz
Installing collected packages: nxppy
  Running setup.py install for nxppy

This last step is blocked ... 

Please help me :(

Thank's a lot !

ejeanneau, 2015-04-12:
It's ok i've downloaded the master version on git and i have execute : 

sudo python setup.py build install

Thank's !

svvitale, 2015-05-11:
Unable to reproduce in 1.3.2 release

closed 2015-05-11 by svvitale