svvitale/nxppy #16: Error while installing nxppy on Raspberry Pi

Loubnar, 2015-07-17:
Hi,

I have encountered a problem while executing the command "sudo python setup.py build install".
It is a "IndexError" and I can't figure out how to fix it. Do you have any idea where this comes from?

Here are the details:

Traceback (most recent call last):
  File "setup.py", line 53, in <module>
    cmdclass = {'build': build_nxppy})
  File "/usr/lib/python2.7/distutils/core.py", line 152, in setup
    dist.run_commands()
  File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "setup.py", line 30, in run
    self.execute(compile, [], 'compiling NxpRdLib')
  File "/usr/lib/python2.7/distutils/cmd.py", line 349, in execute
    util.execute(func, args, msg, dry_run=self.dry_run)
  File "/usr/lib/python2.7/distutils/util.py", line 404, in execute
    func(*args)
  File "setup.py", line 21, in compile
    nxprdlib_dir = glob('neard-explorenfc-*/nxprdlib')[0]
IndexError: list index out of range

I have the latest version of python.

user0007, 2015-07-19:
You have to point on your eard-explorenfc-/nxprdlib folder. I think that using `glob` module in setup.py script is wrong (maybe better solution is to use os.listdir('/') and re.match?)

Loubnar, 2015-07-20:
This fixed the problem!

Thank you very much!

techpet, 2015-09-17:
http://www.nxp.com/redirect/explore-nfc-dev_latest that is referenced in get_nxpRdLib.sh line 46 returns an ERROR 404: Not Found after redirecting to http://www.nxp.com/documents/software/SW282815.zip

svvitale, 2015-09-22:
Yes, confirmed with NXP that the link is fixed, closing.

closed 2015-09-22 by svvitale

eldan84, 2015-12-15:
I got the same problem as techpet on Sep 17. Can I fix this by myself downloading http://cache.nxp.com/documents/software/SW282816.zip ?