svvitale/nxppy #18: set URI + NDEF

marfis89, 2015-08-21:
hi
i want to set the uri identifier (0x00 or 0x01 open website) But with block write i only can set ascii characters?!

marfis89, 2015-08-21:
import nxppy
mifare = nxppy.Mifare()
# Select the first available tag and return the UID
uid = mifare.select()

# Write a URL
mifare.write_block(3, b' ') #SPACE
mifare.write_block(4, b'UID ') # UID BEGIN
mifare.write_block(5, b'elem')
mifare.write_block(6, b'ent1')
mifare.write_block(7, b'4.uk')
mifare.write_block(8, b'UID') # UID END

marfis89, 2015-08-26:
got my own solution with the binascii module. I will post the code in few days

closed 2015-08-26 by marfis89

thibww67, 2020-08-19:
Hi @marfis89,
I know it's a long time ago, but did you remember how to write a NDEF message with this library ? Is it working with write_block ?
Thank you !