svvitale/nxppy #19: Raspberry Pi Error

tuvaergun, 2015-09-07:
[5]+  Stopped                 nano den.py

pi@raspberrypi ~/nxppy $ sudo python den.py
Traceback (most recent call last):
  File "den.py", line 6, in <module>
    uid=mifare.select()
nxppy.SelectError: RequestA command failed: 204

pi@raspberrypi ~/nxppy $ nfc-poll
nfc-poll uses libnfc 1.7.1
NFC reader: pn532_spi:/dev/spidev0.0 opened
NFC device will poll during 30000 ms (20 pollings of 300 ms for 5 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 03  44  
       UID (NFCID1): 04  4f  85  32  a9  37  80  
      SAK (SEL_RES): 20  
                ATS: 75  77  81  02  80  
nfc_initiator_target_is_present: Target Released
Waiting for card removing...done.
pi@raspberrypi ~/nxppy $ 

 Allow device auto-detection (default: true)
# Note: if this auto-detection is disabled, user has to set manually a device
# configuration using file or environment variable
allow_autoscan = true

# Allow intrusive auto-detection (default: false)
# Warning: intrusive auto-detection can seriously disturb other devices
# This option is not recommended, user should prefer to add manually his device.
allow_intrusive_scan = false

# Set log level (default: error)
# Valid log levels are (in order of verbosity): 0 (none), 1 (error), 2 (info), 3 (debug)
# Note: if you compiled with --enable-debug option, the default log level is "debug"
log_level = 1

# Manually set default device (no default)
# To set a default device, you must set both name and connstring for your device
# Note: if autoscan is enabled, default device will be the first device available in devi$
device.name = "Itead_PN532_SPI"
device.connstring = "pn532_spi:/dev/spidev0.0:500000"

svvitale, 2015-09-22:
Hi @tuvaergun, can you verify which NFC board you're using?  I haven't been able to get my board working with libnfc and the configuration you provided.  Is it possible we're using different hardware?  This is the board I'm using: http://www.nxp.com/board/PNEV512R.html

svvitale, 2016-02-15:
Hi @tuvaergun, this issue is nearly 6 months old and I haven't received further details from you.  Closing for now - I'm happy to reopen if you can help me match your configuration/hardware.

closed 2016-02-15 by svvitale