svvitale/nxppy #21: missing setup.py

johnpaddy, 2015-10-09:
I have just received the ExploreNFC board for my Raspberry PI B+. I have installed nxppy and Python 2 doesn't complain so I guess it's installed. During installation the only instruction I failed to get to work was the "sudo python setup.py build install" so obviously I have missed something. No setup.py can be found on the filesystem. Any ideas

johnpaddy, 2015-10-09:
Got the setup.py off this site, but when trying to run the examples the "import nxppy" works but the "mifare=nxppy.Mifare()" causes Python 2.7 to restart.

svvitale, 2015-10-09:
setup.py is only required if you're installing from source.  Are you
installing from source or pip?

johnpaddy, 2015-10-09:
I tried pip first but it didn't work, so then I tried source with the
results above. I will try a clean install on a new card next week and see
if that makes a difference.

svvitale, 2015-10-09:
Yes, if pip doesn't work for some reason, please try to capture a log and
send it over.  That's the preferred installation method unless you're
planning to do development on nxppy itself.  Thanks!

closed 2015-10-09 by svvitale