svvitale/nxppy #22: Installation seemed to work but python program fails

johnpaddy, 2015-10-11:
I made a clean Wheezy installation and used pip to install nxppy. No errors as far as I could see. Trying the python examples gave me restarts at the line "mifare = nxppy.Mifare()". Any ideas? I am probably doing something idiotic but some pointers would be very helpful. Thanks.

johnpaddy, 2015-10-11:
Unable to add log file but the text follows.

pi@raspberrypi ~ $ wget https://bootstrap.pypa.io/get-pip.py
--2015-10-11 09:32:20--  
https://bootstrap.pypa.io/get-pip.py

Resolving bootstrap.pypa.io (bootstrap.pypa.io)... 185.31.18.175

Connecting to bootstrap.pypa.io (bootstrap.pypa.io)|185.31.18.175|:443... 
connected.

HTTP request sent, awaiting response... 200 OK
Length: 1412744 (1.3M) [text/x-python]

Saving to: `get-pip.py'

100%
[===============================================================================================================>]
 1,412,744   1.81M/s   in 0.7s    

2015-10-11 09:32:30 (1.81 MB/s) - `get-pip.py' saved [1412744/1412744]

pi@raspberrypi ~ $ sudo python get-pip.py
Collecting pip
/tmp/tmpPtsmak/pip.zip/pip/_vendor/requests/packages/urllib3/util/ssl_.py:90: 
InsecurePlatformWarning: A true SSLContext object is not available. 
This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. 
For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  
Downloading pip-7.1.2-py2.py3-none-any.whl (1.1MB)
    100% || 1.1MB 18kB/s 
Collecting setuptools
  
Downloading setuptools-18.4-py2.py3-none-any.whl (462kB)
    100% || 462kB 41kB/s 

Collecting wheel
  Downloading wheel-0.26.0-py2.py3-none-any.whl (63kB)
   
 100% | 65kB 190kB/s 

Installing collected packages: pip, setuptools, wheel

Successfully installed pip-7.1.2 setuptools-18.4 wheel-0.26.0
/tmp/tmpPtsmak/pip.zip/pip/_vendor/requests/packages/urllib3/util/ssl_.py:90: 
InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause 
certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.

pi@raspberrypi ~ $ sudo pip install nxppy

Collecting nxppy
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true 

SSLContext object is not available. 
This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. 
For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
 
 InsecurePlatformWarning
  
Downloading nxppy-1.3.2.tar.gz
Building wheels for collected packages: nxppy
  
Running setup.py bdist_wheel for nxppy
  Stored in directory: /root/.cache/pip/wheels/01/ff/f0/9bfba2ae8e84a0fc623fc0e69f7f6db51f9c1d3d17ed02cb40

Successfully built nxppy
Installing collected packages: nxppy
Successfully installed nxppy-1.3.2
pi@raspberrypi ~ $

johnpaddy, 2015-10-15:
Any recommendations for alternative platforms for making an NFC reader? Support for explorenfc seems a bit lacking.

AlterCodex, 2015-10-15:
yes you can try the RC522 I made some test using it and it works

michaelneurohr, 2015-10-16:
Hi johnpaddy,

When using NXPPY you have to tale care that the explore-nfc daemon that comes with the official explore-nfc software is not running in the background.

What about using the official software for the explore-nfc [1]?

[1] http://www.nxp.com/board/PNEV512R.html#documentation

johnpaddy, 2015-10-16:
The official explore nfc software examples succeed in reading some of my
cards but always abort with a segmentation error.
I installed nxppy and tested it before installing the official explorenfc
software so I don't think it is the cause of the problem.

michaelneurohr, 2015-10-16:
Could you please send me an e-mail and describe the problem? That would really help.
You can find my mail address in my profile.

Gadouille, 2015-11-10:
Hi,
I have exactly the same problem has johnpaddy.  I installed nxppy on 2 different raspberry and when I test a simple program it abort after the line "mifare = nxppy.Mifare()" without any notification. I tried with python 2.7 and Python3, the result is the same. Did you find anything intresting to correct the issue?

svvitale, 2015-11-10:
I've still not been able to reproduce this issue.  Can you verify that SPI is enabled and you're running the python script as root (i.e. `sudo test.py`)?

michaelneurohr, 2015-11-10:
@Gadouille Do you have the official software for the EXPLORE-NFC (from the NXP website) and NXPPY installed next to each other? They seem to be conflicting.

Gadouille, 2015-11-12:
Thanks for your answers,
In fact I was trying to test the program with IDLE and IDLE3 and it is not working with it. 
Yet I never tried to directly launch the program in a terminal which is working perfectly well !!!
Thanks for this wrapper, I might now be able to play a little bit with NXP. 

Is it normal that IDLE is not working with it?

svvitale, 2015-11-12:
Were you running IDLE as root (via sudo)?  GPIO access on Raspberry Pi
requires root access by default.

Gadouille, 2015-11-12:
Nop, that's it, I Launched it directly from the operator interface.
If I launch it from the terminal with sudo privilege, it's working.
Thanks again!

closed 2016-02-15 by svvitale