svvitale/nxppy #24: File missing on nxp.com

wkorfhage, 2015-12-08:
The NXP Reader build dies because it can't download the file. Browsing the explore-nfc documents on the NXP site, it appears the file is gone. I see files SW282811, SW282815, SW282714 and SW282715. SW282815 says it is Rev 1.5 had has the sources of all the components. Has the source been updated?

    [Downloading NXP Reader Library]
    --2015-12-07 20:04:46--  http://www.nxp.com/redirect/explore-nfc-dev_latest
    Resolving www.nxp.com (www.nxp.com)... 23.197.26.138
    Connecting to www.nxp.com (www.nxp.com)|23.197.26.138|:80... connected.
    HTTP request sent, awaiting response... 302 Moved Temporarily
    Location: http://www.nxp.com/documents/software/SW282814.zip [following]
    --2015-12-07 20:04:46--  http://www.nxp.com/documents/software/SW282814.zip
    Reusing existing connection to www.nxp.com:80.
    HTTP request sent, awaiting response... 302 Moved Temporarily
    Location: http://www.nxp.com/documents/application_note/SW282814.zip [following]
    --2015-12-07 20:04:46--  http://www.nxp.com/documents/application_note/SW282814.zip
    Reusing existing connection to www.nxp.com:80.
    HTTP request sent, awaiting response... 404 Not Found
    2015-12-07 20:04:46 ERROR 404: Not Found.

newpipe, 2015-12-08:
Yesterday the NXP and Freescale websites got merged. Looks like the file needed for NXPPY is missing now. I'll take this up as soon as possible to get the file back onto the website.

Thanks for reporting,
Michael

grubis, 2015-12-10:
Same issue with my build try today... thanks for looking at this newpipe, looking forward to building off your great work.

AlterCodex, 2015-12-10:
I'm having the same issue, there is any other repository to download the zips files?

michaelneurohr, 2015-12-11:
I'm sorry the fix takes so long on our side. My colleagues are still working on it.
Now I made the file temporary available for you at
https://nxp.box.com/shared/static/t0cs19rvveogcsbmb094sdey3l1tyeir.zip

Just change the URL in the Python script.

zapito, 2015-12-11:
Hi all,
Same issue for me. I've some questions where surely you could help me (first I've to apologize as I'm totally newbie installing libraries and also with the NXP Explorer card):

I'm running everything in a Raspberry Pi 2 (Jessie) and I already installed the example software from NXP (SW282715.zip). Everything works fine with the example (-$ explorenfc-basic) and I can detect different NFC chips from several manufacturers. Now coming back to nxppy:

1.- Should I de-install the (.deb) packages already installed, as pip3 install nxppy will try to compile everything again from the sources?, could it cause some kind of conflict if wiringpi, etc. is already present in the system?.
2.- As it is my first attempt to do some project I was thinking about installing the nxppy in a virtual environment ([myenv]user:Host$ pip3 install nxppy) to avoid any conflict with the rest of phyton libraries. Is it possible, right?
3.- @michaelneurohr: sorry for this third stupid question, but after changing the URL in the file get_nxpRdLib.sh, how can I make the installation from my computer (downloading nxppy-master.zip and modifying the affected file)?.

Again sorry for my questions and thank you for your patient.

Regards.

grubis, 2015-12-11:
Thanks for the work-around... I modified the shell script to copy from my local download directory instead of fetch and everything built and installed correctly. I'm reading and writing tags now!

michaelneurohr, 2015-12-11:
@zapito: Yes, please remove the explore-nfc package if you want to use nxppy otherwise you will get some conflicts. You can keep WiringPi installed.

After downloading the master branch contained in a zip file, just unzip it, modify the shell script to download the NFC Reader LIbrary from NXP with the new URL and run setup.py.

zapito, 2015-12-11:
@michaelneurohr: Thanks a lot, now it works. I follow your hints and finish with: $sudo python setup.py build install.

Regards.

forsanta, 2015-12-15:
hello @michaelneurohr 
I've the same issue, can you make a other tempory file in download ?

Regards.
ps : sorry for my english, i'm french ....

AlterCodex, 2015-12-15:
In my branch is the  zip file 
Att
Altercodex

forsanta, 2015-12-15:
thank

just to be sure , this branch https://github.com/AlterCodex/nxppy/tree/MifareClassic ?

regard

AlterCodex, 2015-12-15:
Yes, but you need to change the script

forsanta, 2015-12-15:
ok i get the 
SW282816.zip and 
the get_nxpRdLib.sh  in the branch for the classic and the other file in the master branch

svvitale, 2015-12-15:
@michaelneurohr any idea when this will be resolved on nxp.com?  @AlterCodex, unfortunately NXP does not allow these files to be stored on GitHub.  Can you please remove the file from your branch?

AlterCodex, 2015-12-15:
Im right now out of the office I will remove it this afternoon.

michaelneurohr, 2015-12-16:
The URL is working again. It is resolved now since about 30 minutes.
I'm very sorry for the troubles you had.

AlterCodex, 2015-12-16:
dear @michaelneurohr  and @svvitale  the file was removed from my branch but I will like to chat with some one from nxp its that possible?.

svvitale, 2016-02-15:
Verified today that the URL is working as well as installation via pip.  Closing.

closed 2016-02-15 by svvitale