svvitale/nxppy #26: I2C get blocked

AlterCodex, 2016-01-18:
I'm trying to use the explorer module via spi and a rtc using i2c but when I run the py file that uses the nxp explorer the i2c get blocked.

svvitale, 2016-02-15:
@michaelneurohr does the explorer utilize the i2c pins at all?  Can you think of why it would be blocking?

michaelneurohr, 2016-02-15:
> @michaelneurohr[1] does the explorer utilize the i2c pins at all?

No, it uses the SPI interface only. I'll check in the NFC Reader  
Library why the I2C pins are getting blocked.

Michael

Links:
------
[1] https://github.com/michaelneurohr
[2] https://github.com/svvitale/nxppy/issues/26#issuecomment-184068064

svvitale, 2016-03-19:
@AlterCodex it sounds like this is being fixed in an upcoming release of the NXP Reader Library.  I'll update this ticket when those changes become available.

kevin7979, 2016-03-27:
Hi, 
Does this problem fix yet? I have same problem that it block all my I2C after i run the NXP explore's py file.

AlterCodex, 2016-03-28:
It is not a problem of the nxppy you can modify the cop files of the box controller to avoid this problem in the official forum of the NXP explorer you can find some temporal solution

svvitale, 2016-06-15:
@AlterCodex, the latest release (1.4.0) uses the newer NXP NFC Reader Library which should no longer block I2C.  Closing this issue for now, please re-open if you still encounter issues after upgrading.

closed 2016-06-15 by svvitale