svvitale/nxppy #28: raspberry pi with ubuntu mate error

lmbn, 2016-02-12:
I try to install it on a raspberry pi with ubuntu mate 15.10. These are the steps i follow:

sudo apt-get update
sudo apt-get install python2.7-dev
sudo apt-get install cmake
sudo apt-get install python-pip
sudo apt-get install git
sudo apt-get install nxppy

And I get this error:

neard-explorenfc-0.9/nxprdlib/libnxprdlib.a: error adding symbols: Bad value

collect2: error: ld returned 1 exit status

error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

----------------------------------------
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-Zzb__5/nxppy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-8TRpjA-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-Zzb__5/nxppy
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 248, in main
    return command.main(cmd_args)
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 161, in main
    text = '\n'.join(complete_log)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 32: ordinal not in range(128)

Any ides? Thank you!

lmbn, 2016-02-12:
then i try to install from source and I get this error:

In file included from Mifare.c:1:0:
Mifare.h:31:25: fatal error: ph_NxpBuild.h: No such file or directory
compilation terminated.
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

svvitale, 2016-02-15:
Hadn't tried Mate yet, so I loaded it up this evening.  A couple things were broken - first the version of pip was causing issues.  Start by installing the latest version of pip:

```
wget https://bootstrap.pypa.io/get-pip.py
sudo python get-pip.py
```

Then I discovered that Mate uses an ARMv7 compiler that needed a few adjustments to its optimization settings and linker arguments.  I committed those changes under ce32754.  You can install them directly with this command:

```
sudo pip install git+git://github.com/svvitale/nxppy.git@ce32754
```

Let me know if that doesn't work for you.  Thanks for the bug report.

closed 2016-02-15 by svvitale

lmbn, 2016-02-20:
Thank you! It working now!

Cognaiscance, 2017-04-29:
This fix also worked for mine, thx.