svvitale/nxppy #29: Use of pins ("This channel is already in use")

wingesas, 2016-03-10:
Hi there,

I' m using nxppy to playback songs in mopidy (mpd). This works great.

Since the nfc card does not use all the pins [(see here)](https://hackaday.io/project/1812-ourticket/log/5159-explore-nfc-layout), I connected three buttons to the pins 7, 8, 13 to get more functions (play next, prev, pause). This also works for pin 7, but for 8 and 13 I get the message "This channel is already in use" and the pins switched from high to low once I do "mifare.select()".

I'm not sure if this is an issue in nxppy or the underlying nfc software, but it seems like that pins are used, which are not bound to the card.

Cheers,
Sascha

svvitale, 2016-03-10:
@michaelneurohr can you give any insight here?  @wingesas, the nxppy library makes use of NXP's reader library from a very high level.  We don't have any visibility into what's happening on the physical pins, we rely on the library for that low-level control.  Hopefully Michael can help answer your question.  Thanks!

wingesas, 2016-03-13:
Hi Scott,

thanks for the information. It worked for me using the pins 7, 8 and 10. [Here](https://github.com/wingesas/nfc-playlist/blob/master/nfc-playlist/nfcPlaylist.py) is my code, for anyone who is interested.
I'm closing this, since it is not an issue for nxppy.

Cheers,
Sascha

closed 2016-03-13 by wingesas