svvitale/nxppy #30: can't work to show NFC ID

helxsz, 2016-03-11:
I am using the EXPLORE-NFC with PI, I have enabled the SPI on the PI, and used the example polling example, but when the NFC approached to the NFC reader, no print out, and nothing happens.

Here is the picture of 
https://www.dropbox.com/s/4pw10m2ztucav77/Screenshot%202016-03-11%2007.53.25.png?dl=0
https://www.dropbox.com/s/0nxuwrv83481srq/Screenshot%202016-03-11%2007.53.35.png?dl=0

can you give me a inspiration where goes wrong ?

michaelneurohr, 2016-03-11:
I assume, you are using the software from [1]? Please note, that NXPPY can not be used at the same time you use the EXPLORE-NFC software is running its daemon in the background.

[1] http://www.nxp.com/products/identification-and-security/nfc-and-reader-ics/nfc-frontend-solutions/explore-nfc-exclusive-from-element14:PNEV512R?fpsp=1&tab=Design_Tools_Tab

svvitale, 2016-03-19:
@helxsz please feel free to reopen this if @michaelneurohr's suggestion did not resolve your issue.

closed 2016-03-19 by svvitale