svvitale/nxppy #31: How to read nfc data

zicheee, 2016-03-15:
Hello,

I try to read data from a nfc tag with python.
I test the example from the readme.md.
How do i read the data on the tag? 
When trying 
`data = mifare.read_block(0)
print(data)`
I get something like "ðn� "€ƒHá"

With my Android App i see the following:
Dataset 0: text/plain 
"myText"

How can i read this content?

michaelneurohr, 2016-03-17:
It looks like you are starting to read directly from the beginning of the memory. But, depending on the tag you are using, the NDEF message does not start at the beginning of the memory. At the beginning you always have some manufacturing information like the date, the UID and others. On a NTAG for example, the NDEF message starts from page 4. See the datasheet of the NTAG [1], page 11. But the letters of the NDEF message itself start around page 7, because before you have the NDEF header.
To see the complete memory mapping and content of your tag, you could use the Android application NXP TagInfo [2]. This application helped me already a lot during development.

[1] http://cache.nxp.com/documents/data_sheet/NTAG213_215_216.pdf
[2] https://play.google.com/store/apps/details?id=com.nxp.taginfolite

zicheee, 2016-03-18:
Thank you for your reply,
But i don't get it how it should work. From NXP-Taginfo I know it's an "NXP MIFARE Ultralight NTAG203".
I create a dataset "e.g. text/plain: this is my string i saved here" and write it to the tag.
Example Code:

`uid = mifare.select()`
`data = mifare.read_block(7)`
`print("UID: %s" % uid)`
`print(data)`

Output:
`UID: 04F06E81A02280`
`enthis is my str`

I tried read_block(4), 5, 6, 7, 8 etc. I have no idea how to determine which data is on the tag (eg. url, plain text etc.) and how to get exactly this data and not only a part of it

michaelneurohr, 2016-03-18:
NFC Forum specifies a so called NDEF Detection Procedure for every supported Tag Type. For Type 2 Tags, like you are using, you can find this in the document NFC Forum Type 2 Tag Operation Specification 1.2 in chapter 6.4.1.
If you are wondering how you can differentiate between the different NDEF types, you can find the information in the document NFC Data Exchange Format (NDEF) Technical Specification.

So in short: handling NDEF messages without any third party library means you have to read the according specifications and implement accordingly. But I guess, this is a bit out of scope for you. Maybe you can find an NDEF library on the internet.

But why don't you just use the official support software for the EXPLORE-NFC [1]? This software does already implement all these functionalities. This software already comes, additionally to the source code, with examples in binary form, which can be used right away for writing and reading NDEF messages. You could call these binaries and simply parse the output directly from your Python program.

[1] http://www.nxp.com/products/identification-and-security/nfc-and-reader-ics/nfc-frontend-solutions/explore-nfc-exclusive-from-element14:PNEV512R?fpsp=1&tab=Design_Tools_Tab

zicheee, 2016-03-18:
Good idea :-)
I never thought about that easy way. Working like i expected

closed 2016-03-18 by zicheee