svvitale/nxppy #33: Latest version of neard-explorenfc seems to lack ./nxprdlib/libnxprdlib.a

LucardoNL, 2016-03-22:
Hey there,

While trying to install nxppy on my Raspi3 I kept running into several issues, which I've narrowed down to problem as stated in the subject. I hit this point after manually following all the download links from get_nxpRdLib.sh. WiringPi was also an issue where the build couldn't clone the repo from it's originally stated source, but changing the URL there to the github url fixed that issue.

Now my build ends in the following error:

`creating build/lib.linux-armv7l-2.7
arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security build/temp.linux-armv7l-2.7/Mifare.o build/temp.linux-armv7l-2.7/nxppy.o -o build/lib.linux-armv7l-2.7/nxppy.so neard-explorenfc-0.9/nxprdlib/libnxprdlib.a -lwiringPi
arm-linux-gnueabihf-gcc: error: neard-explorenfc-0.9/nxprdlib/libnxprdlib.a: No such file or directory
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
`

Manually checking the latest neard-explorenfc-0.9 folder it seems the file libnxprdlib.a is indeed missing. Any ideas?

svvitale, 2016-03-23:
Hi @LucardoNL,

This file is built as part of the install process, it doesn't come from the
neard-explorenfc zip file.  Unfortunately I don't yet have a Pi3 to test
on.  Are you running Raspian?  Can you try pulling the nxppy source and
then running `sudo ./get_nxpRdLib.sh` manually?

LucardoNL, 2016-03-23:
Hey @svvitale,

Firstly, thank you very much for your time to answer my question!

Running sudo ./get_nxpRdLib.sh ends with no errors. What would my next step then be?

LucardoNL, 2016-03-24:
I'n not quite sure how, but I got it to work! 

I ran the ./bootstrap, configure, make commands etc for the neard-explorer-* package manually following its readme, then retried the setup.py build install and this time it finished. 

Thanks again for the reply.

closed 2016-03-24 by LucardoNL