svvitale/nxppy #34: Problem Installing, unrecognized command line option ‘-fstack-protector-strong’

matthieuvisian, 2016-04-27:
Hi,

Trying to install nxppy on Raspberry 2 this error occurs:
All goes well untill :

```
running build_ext
building 'nxppy' extension
creating build
creating build/temp.linux-armv7l-2.7
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Ineard-explorenfc-0.9/nxprdlib/types -Ineard-explorenfc-0.9/nxprdlib/intfs -I/usr/include/python2.7 -c Mifare.c -o build/temp.linux-armv7l-2.7/Mifare.o -O1
cc1: error: unrecognized command line option ‘-fstack-protector-strong’
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```
I did some research but can't find anything.

I need some help.

Thanks in advance !

matthieuvisian, 2016-04-27:
It seems : arm-linux-gnueabihf-gcc is an gcc 4.9 option.

And my gcc was 4.8 version. I updated it but I got still that error

svvitale, 2016-04-28:
Which OS distribution and version are you using?  I'll see if I can
reproduce this.  That part of the command line is put together by the
Python distutils, so there may be some version incompatibility there.
Also, which version of python?

matthieuvisian, 2016-04-29:
Hi again,

I'm using Raspberry PI 2 with the latest version of Jessie.

Python version : 2.7.9

svvitale, 2016-06-15:
This should be resolved in the latest release (1.4.0).  Closing this issue for now, please re-open if you still encounter issues after upgrading.  Note that I haven't done any recent testing on Wheezy, only Jessie.

closed 2016-06-15 by svvitale