svvitale/nxppy #37: Can't install - IndexError: list index out of range

bottlefury, 2016-05-12:
Hi... I know others have had this problem but i'm not clear what the solution is (am a vb.net programmer so finding my way with Python/ Linux).  Now the weird thing is I did manage to install and use Nxppy a few weeks ago but have since had to start again for PI related reasons... now it seems absolutely impossible to install as I continually get the nxprdlib_dir = glob('neard-explorenfc-/nxprdlib')[0]
IndexError: list index out of range...

I have seen some mention of pointing an installation file to a new location but am just not sure what to do...HELP lol

yon2004, 2016-05-29:
could it be because you do not have git installed?
sudo apt-get install git

svvitale, 2016-06-15:
This has been resolved in the latest release (1.4.0).  Closing this issue for now, please re-open if you still encounter issues after upgrading.

closed 2016-06-15 by svvitale