svvitale/nxppy #42: nxppy.SelectError: Card detected: 4084

RasPinator, 2016-07-26:
Hi there, 

I'm having trouble getting a simple test python app to run 
on a Raspberry Pi 2 Model B running Raspbian jessie:

```
import nxppy

mifare = nxppy.Mifare()
uid = mifare.select()
print(uid)
```

I keep getting an Error at:

`uid = mifare.select()`

which says:

`nxppy.SelectError: No Card detected: 4084`

whether or not a tag is placed on the nxp-module.

After failing quite a few times I tried reading the tag with the explore-nfc basic tool and that worked...

svvitale, 2016-07-26:
Can you try it in a loop just so you can be sure the card is within the field?

```
import nxppy
import time

mifare = nxppy.Mifare()

while True:
    try:
        uid = mifare.select()
        print(uid)
    except:
        print("No card found")
        time.sleep(1)
```

Also, what version of nxppy do you have installed?  In older versions, you could not use the explore-nfc basic tool and nxppy at the same time.  Make sure that the explore-nfc daemon is not running before trying to run the python version.

RasPinator, 2016-07-27:
Thank you very much for your answer!

Unfortunately I already tried it both ways (with and without the loop).
I installed nxppy via 

`sudo pip install nxppy`

and it is version 1.4.3
Also, I did try to run it with and without the explore-nfc deamon being installed.

svvitale, 2016-07-27:
I forgot to ask, is this on Raspbian?  What version?  I'll see if I can
reproduce the issue.

RasPinator, 2016-07-27:
It's Raspbian Jessie Version May 2016

svvitale, 2016-07-28:
I've seen some rumblings about that release having SPI issues.  Do you have
an older version handy to try?  I'll try to look at this myself, but my
travel schedule is currently a little out of hand.

RasPinator, 2016-07-29:
Thank you very much!!! 
It works with an older Version (Linux raspberrypi 4.1.13-v7+ #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015 armv7l GNU/Linux)

svvitale, 2016-07-29:
Glad to hear!  I'll leave this ticket open until there's a new release to test against.  It's possible something was purposely changed and inadvertently broke nxppy, but I'm willing to wait 1 release to verify.

svvitale, 2016-09-05:
I reinstalled the latest Raspian today and successfully tested nxppy on both Python 2.7 and 3.4.  The only change needed was to enable SPI.  Not sure what the original problem was, but glad to hear things are working, @RasPinator.  Closing this ticket.

closed 2016-09-05 by svvitale