svvitale/nxppy #43: nxppy.WriteError: Write failed: 706

joswr1ght, 2016-07-31:
With several different Mifare UltraLight cards, I'm unable to write to the card in a consistent manner. I keep getting ```nxppy.WriteError: Write failed: 706``` errors:

```
pi@raspberrypi:~ $ cat testwrite.py
import nxppy, sys
mifare = nxppy.Mifare()
randfile = open("/dev/urandom")
# Print card UIDs as they are detected
while True:
    try:
        uid = mifare.select()
        print(uid)
        print "Writing to blocks:",
        for i in range(4,15):
            print i,
            mifare.write_block(i, randfile.read(4))
    except nxppy.SelectError:
        # SelectError is raised if no card is in the field.
        pass
    print
    sys.exit(0)
```

The Write failed 706 error doesn't always happen when writing to the same block, and occasionally it will write all the block correctly:

```
pi@raspberrypi:~ $ python testwrite.py
046F65AA9B3280
Writing to blocks: 4 5
Traceback (most recent call last):
  File "testwrite.py", line 12, in <module>
    mifare.write_block(i, randfile.read(4))
nxppy.WriteError: Write failed: 706
pi@raspberrypi:~ $ python testwrite.py
046F65AA9B3280
Writing to blocks: 4
Traceback (most recent call last):
  File "testwrite.py", line 12, in <module>
    mifare.write_block(i, randfile.read(4))
nxppy.WriteError: Write failed: 706
pi@raspberrypi:~ $ python testwrite.py
046F65AA9B3280
Writing to blocks: 4 5 6 7 8 9 10 11 12 13 14
pi@raspberrypi:~ $ python testwrite.py
046F65AA9B3280
Writing to blocks: 4 5 6
Traceback (most recent call last):
  File "testwrite.py", line 12, in <module>
    mifare.write_block(i, randfile.read(4))
nxppy.WriteError: Write failed: 706
```

Thoughts on why this write failed error is happening? I'm running a Raspberry Pi 3 with Raspbian 8.0. Thanks!

svvitale, 2016-08-19:
Hi @joswr1ght,

Looking through the NXP reader library's error codes, this appears to be a protocol error in the Mifare PAL component.  Since it works sometimes, I would guess that it's failing due to intermittent RF issues.  For the code you're writing, would it make sense to add a retry mechanism?

The NXP reader library is pretty low-level, so it doesn't do much error handling or retrying.  Those functions are left to the caller.

Hope this helps.  Closing out the issue for now.  Let me know if there's something you'd like me to look into further.  Thanks!

-Scott

closed 2016-08-19 by svvitale

joswr1ght, 2016-08-19:
Thanks for looking into this @svvitale. -Josh