svvitale/nxppy #45: Install issue : Failed building wheel for nxppy

Blablach, 2016-09-20:
Hi, 
I've tried to install nxppy but i't return error..
I've tried with `pip install nxppy` and tried to clone the git and run `python setup.py build install` but they both return the error :

_Failed building wheel for nxppy_ 
_Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-pV6OJF/nxppy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-dHyM_9-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-pV6OJF/nxppy/_

I've tried on 3 different devices but all crash..

Ps: sorry for my english ;)

svvitale, 2016-09-20:
Are you trying to install into a virtualenv or into your global Python library?  If the latter, you'll want to run `sudo pip install nxppy`.

svvitale, 2016-10-17:
Closing for now. Please let me know if you're still having issues.

closed 2016-10-17 by svvitale