svvitale/nxppy #52: Can't read from various tags

DamienMolina, 2017-01-30:
Hello,
I try to read from some tags:
* The NFC sample card delivered with my explore-nfc-ww reader.
* A Sony Xperia Z3 compact.
* A M24SR64 from ST

I can detect the tag, and read the UUID for all 3 of them.
However I can only read the Sample card, and not the two others. I get this error:
`nxppy.ReadError: Read failed: 0201`
What is the meaning of the error code 0201?
Is it due to the tag type?

svvitale, 2017-02-01:
Hi @DamienMolina,

0x201 indicates that there was a timeout in the HAL component.  It's required that you activate a tag using `select()` first, and then proceed to read data.  Have you tried that?  Example code below.

```python
import nxppy
m = nxppy.Mifare()
m.select()
m.read_block(0)
```

svvitale, 2017-02-19:
@DamienMolina, let me know if you need additional help here and I'll reopen the issue.

DamienMolina, 2017-02-20:
Sorry, I didn't have access to my NFC Reader.
Here's my test with the select():
```>>> uid = m.select()
>>> uid
'028400176045FF'
>>> m.read_block(10)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
nxppy.ReadError: Read failed: 0706
>>> m.read_block(0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
nxppy.ReadError: Read failed: 0706
```

svvitale, 2017-02-20:
0706 indicates that the Mifare component of the NXP reader library encountered a response that violates the Mifare protocol.  Are you sure this is a Mifare-compatible tag?

DamienMolina, 2017-02-23:
I think you're right, I don't have a Mifare-compatible tag, it comes from ST. I thought it could be all right nonetheless.

closed 2017-02-23 by DamienMolina