svvitale/nxppy #55: phhwConfig.h not found

NickvDr, 2017-03-09:
Hi,

I'm trying to install your nxppy with python 2.7.9 on Raspbian Jessie but got the following error:

`sudo pip install git+git://github.com/svvitale/nxppy.git
Collecting git+git://github.com/svvitale/nxppy.git
  Cloning git://github.com/svvitale/nxppy.git to /tmp/pip-tJWzqS-build
Installing collected packages: nxppy
  Running setup.py install for nxppy ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-tJWzqS-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-BYgeuc-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_ext
    compiling NxpRdLib
    [Downloading NXP Reader Library]
    --2017-03-09 13:48:24--  https://nxp.box.com/shared/static/xxuwpzh2ztl63b8ujsclg5wqsmlxfmw1.zip
    Resolving nxp.box.com (nxp.box.com)... 107.152.26.197, 107.152.27.197
    Connecting to nxp.box.com (nxp.box.com)|107.152.26.197|:443... connected.
    HTTP request sent, awaiting response... 302 Found
    Location: https://nxp.app.box.com/shared/static/xxuwpzh2ztl63b8ujsclg5wqsmlxfmw1.zip [following]
    --2017-03-09 13:48:24--  https://nxp.app.box.com/shared/static/xxuwpzh2ztl63b8ujsclg5wqsmlxfmw1.zip
    Resolving nxp.app.box.com (nxp.app.box.com)... 107.152.27.199
    Connecting to nxp.app.box.com (nxp.app.box.com)|107.152.27.199|:443... connected.
    HTTP request sent, awaiting response... 404 Not Found
    2017-03-09 13:48:25 ERROR 404: Not Found.
   
    [nxp.zip]
      End-of-central-directory signature not found.  Either this file is not
      a zipfile, or it constitutes one disk of a multi-part archive.  In the
      latter case the central directory and zipfile comment will be found on
      the last disk(s) of this archive.
    unzip:  cannot find zipfile directory in one of nxp.zip or
            nxp.zip.zip, and cannot find nxp.zip.ZIP, period.
    ./get_nxpRdLib.sh: line 17: pushd: nxp/build: No such file or directory
    CMake Error: The source directory "/tmp" does not appear to contain CMakeLists.txt.
    Specify --help for usage, or press the help button on the CMake GUI.
    make: *** No rule to make target 'NxpRdLibLinuxPN512'.  Stop.
    ./get_nxpRdLib.sh: line 20: popd: directory stack empty
    building 'nxppy' extension
    creating build
    creating build/temp.linux-armv7l-2.7
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DLINUX -DNATIVE_C_CODE -DNXPBUILD_CUSTOMER_HEADER_INCLUDED -DNXPBUILD__PHHAL_HW_RC523 -I/usr/include/python2.7 -c Mifare.c -o build/temp.linux-armv7l-2.7/Mifare.o -O0 -std=gnu99 -isystemnxp/nxprdlib/NxpRdLib/intfs -isystemnxp/nxprdlib/NxpRdLib/types -isystemnxp/nxprdlib/NxpRdLib -isystemnxp/linux/intfs -isystemnxp/linux/comps/phbalReg/src/Linux -isystemnxp/linux/shared -isystemnxp/examples/NfcrdlibEx4_MIFAREClassic/intfs -isystemnxp/nxprdlib/NxpRdLib/comps/phbalReg/src/Stub -isystemnxp/linux/comps/phPlatform/src/Posix -isystemnxp/linux/comps/phOsal/src/Posix
    In file included from Mifare.c:1:0:
    Mifare.h:32:24: fatal error: phhwConfig.h: No such file or directory
     #include <phhwConfig.h>
                            ^
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
   
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-tJWzqS-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-BYgeuc-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-tJWzqS-build/`

Could you please help?

Thx!

ahilti, 2017-03-09:
Experiencing the same issue but it's due to
https://github.com/svvitale/nxppy/blob/master/get_nxpRdLib.sh#L11
in get_nxpRdLib.sh pointing to a non existing Explore NFC Library File.

NickvDr, 2017-03-09:
@ahilti Thanks for the quick reply!

Is there already a fix for this?

NickvDr, 2017-03-09:
I found a fix for it:

Download the following zip: https://cache.nxp.com/secured/assets/downloads/data/en/software/SW3693-1.zip?__gda__=1489075520_224a1c84b368c9877839e4bc078e3224&fileExt=.zip

Then rename it to nxp.zip and store it somewhere online.

After storing it online edit https://github.com/svvitale/nxppy/blob/master/get_nxpRdLib.sh#L11 and replace the current link for your link.

That would do the trick.

michaelneurohr, 2017-03-09:
Please try again with the commit d30583d82b4c93c5e9a321591d4e2adaa4eb75c4

svvitale, 2017-03-14:
Verified broken state, and verified correct operation after @michaelneurohr's commit.  Thanks!

cegea, 2018-01-15:
Hi,

The link has changed again. [https://github.com/svvitale/nxppy/blob/master/get_nxpRdLib.sh#L11](https://github.com/svvitale/nxppy/blob/master/get_nxpRdLib.sh#L11)

svvitale, 2018-01-15:
@michaelneurohr can you provide an updated link?

cegea, 2018-01-15:
I can't, sorry. I think the problem is that you need to be logged to download the file each time, so the link won't last a day.

michox, 2018-01-16:
can someone provide a link or that zip file? i really need it

cegea, 2018-01-16:
Look for sw3693. [NXP engine search](https://www.nxp.com/search?client=nxp_search_all_results&site=nxp_en&proxystylesheet=nxp_search_style_fe&sort=date%3AD%3AL%3Ad1&oe=UTF-8&ie=UTF-8&ud=1&output=xml_no_dtd&exclude_apps=1&callback=ss_show&lang_cd=en&filter=0&getfields=*&baseUrl=%2F%2Fwww.nxp.com%2Fwebapp&SEARCH_OPERATOR=Contains&attempt=-1&rc=1&hl=en&dnavs=&q=sw3693)

rsydwidiadi, 2018-01-21:
What is the next step after download it? Sorry I still don't understand how it works.

michox, 2018-01-21:
after you download the archive, you rename it to nxp.zip and then upload it somewhere. I uploaded it to dropbox with the link https://www.dropbox.com/s/yrl2hqvda1r07lx/nxp.zip?dl=1
You can try that one but im not sure if dropbox might have issues with downloading something over wget, I used a dropbox command instead of wget. If it doesn't work find an upload service upload it yourself and try it with that link
Next you open the get_nxpRdLib.sh with a text editor and change the link at line 11 to mine or yours. after that run `sudo python setup.py install` in the root directory of your git clone

svvitale, 2018-01-21:
Hi all, I've just committed d47ef72 which will look for a local file named `nxp.zip` before trying (and failing) to download it directly from NXP.  I'm working with the NXP representatives to figure out the best way to manage this going forward.

For now, you can clone the nxppy master branch, download the reader library directly from NXP and save as nxp.zip.  Then run `python setup.py install` to install nxppy.

@michox, unfortunately their licensing does not allow for hosting this source code on an external website, which is why we're pulling it from them directly.

sabberlatz, 2018-01-25:
I still have difficulties, installing nxppy. 

here the logfile:

pi@raspberrypi:~ $ sudo python setup.py install
running install
running bdist_egg
running egg_info
creating nxppy.egg-info
writing nxppy.egg-info/PKG-INFO
writing top-level names to nxppy.egg-info/top_level.txt
writing dependency_links to nxppy.egg-info/dependency_links.txt
writing manifest file 'nxppy.egg-info/SOURCES.txt'
reading manifest file 'nxppy.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching '*.sh'
writing manifest file 'nxppy.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_ext
compiling NxpRdLib
/bin/sh: 1: ./get_nxpRdLib.sh: not found
building 'nxppy' extension
creating build
creating build/temp.linux-armv7l-2.7
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DLINUX -DNATIVE_C_CODE -DNXPBUILD_CUSTOMER_HEADER_INCLUDED -DNXPBUILD__PHHAL_HW_RC523 -I/usr/include/python2.7 -c Mifare.c -o build/temp.linux-armv7l-2.7/Mifare.o -O0 -std=gnu99 -isystemnxp/nxprdlib/NxpRdLib/intfs -isystemnxp/nxprdlib/NxpRdLib/types -isystemnxp/nxprdlib/NxpRdLib -isystemnxp/linux/intfs -isystemnxp/linux/comps/phbalReg/src/Linux -isystemnxp/linux/shared -isystemnxp/examples/NfcrdlibEx4_MIFAREClassic/intfs -isystemnxp/nxprdlib/NxpRdLib/comps/phbalReg/src/Stub -isystemnxp/linux/comps/phPlatform/src/Posix -isystemnxp/linux/comps/phOsal/src/Posix
In file included from Mifare.c:1:0:
Mifare.h:32:24: fatal error: phhwConfig.h: No such file or directory
 #include <phhwConfig.h>
                        ^
compilation terminated.
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

sabberlatz, 2018-01-25:
....and if I download the nxp.zip file from the link, provided in the .sh file, it has only 22kb and is corrupt. 

any idea?

should be so easy.....

sabberlatz, 2018-01-25:
I have still difficulties. If I try to compile the stuff I receive at 87% following error:

========================================
[ 85%] Building C object NfcrdlibEx3_NFCForum/CMakeFiles/NfcrdlibEx3_NFCForum.dir/src/phApp_Init.c.o
[ 85%] Linking C executable NfcrdlibEx3_NFCForum
[ 85%] Built target NfcrdlibEx3_NFCForum
Scanning dependencies of target NfcrdlibEx4_MIFAREClassic
[ 86%] Building C object NfcrdlibEx4_MIFAREClassic/CMakeFiles/NfcrdlibEx4_MIFAREClassic.dir/NfcrdlibEx4_MIFAREClassic.c.o
[ 86%] Building C object NfcrdlibEx4_MIFAREClassic/CMakeFiles/NfcrdlibEx4_MIFAREClassic.dir/src/phApp_Init.c.o
[ 87%] Linking C executable NfcrdlibEx4_MIFAREClassic
[ 87%] Built target NfcrdlibEx4_MIFAREClassic
Scanning dependencies of target NfcrdlibEx5_ISO15693
[ 87%] Building C object NfcrdlibEx5_ISO15693/CMakeFiles/NfcrdlibEx5_ISO15693.dir/NfcrdlibEx5_ISO15693.c.o
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c: In function ‘NfcrdlibEx5_ISO15693’:
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:209:43: error: ‘phacDiscLoop_Sw_DataParams_t’ has no member named ‘sTypeVTargetInfo’
                 phApp_Print_Buff(sDiscLoop.sTypeVTargetInfo.aTypeV[0].aUid, 0x08);
                                           ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:212:47: error: ‘phacDiscLoop_Sw_DataParams_t’ has no member named ‘sTypeVTargetInfo’
                 memcpy(aReceivedUid, sDiscLoop.sTypeVTargetInfo.aTypeV[0].aUid, 0x08);
                                               ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:215:50: error: ‘phacDiscLoop_Sw_DataParams_t’ has no member named ‘sTypeVTargetInfo’
                 if (DisplayCardTypeInfo(sDiscLoop.sTypeVTargetInfo.aTypeV[0].aUid) == PH_ERR_SUCCESS)
                                                  ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:226:62: error: ‘salI15693’ undeclared (first use in this function)
                         status = phalI15693_ReadSingleBlock(&salI15693,
                                                              ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:226:62: note: each undeclared identifier is reported only once for each function it appears in
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:227:29: error: ‘PHAL_I15693_OPTION_OFF’ undeclared (first use in this function)
                             PHAL_I15693_OPTION_OFF,
                             ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:286:26: error: ‘PHPAL_SLI15693_FLAG_NBSLOTS’ undeclared (first use in this function)
                         (PHPAL_SLI15693_FLAG_NBSLOTS | PHPAL_SLI15693_FLAG_DATA_RATE | PHPAL_SLI15693_FLAG_INVENTORY),
                          ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:286:56: error: ‘PHPAL_SLI15693_FLAG_DATA_RATE’ undeclared (first use in this function)
                         (PHPAL_SLI15693_FLAG_NBSLOTS | PHPAL_SLI15693_FLAG_DATA_RATE | PHPAL_SLI15693_FLAG_INVENTORY),
                                                        ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:286:88: error: ‘PHPAL_SLI15693_FLAG_INVENTORY’ undeclared (first use in this function)
                         (PHPAL_SLI15693_FLAG_NBSLOTS | PHPAL_SLI15693_FLAG_DATA_RATE | PHPAL_SLI15693_FLAG_INVENTORY),
                                                                                        ^
NfcrdlibEx5_ISO15693/CMakeFiles/NfcrdlibEx5_ISO15693.dir/build.make:62: recipe for target 'NfcrdlibEx5_ISO15693/CMakeFiles/NfcrdlibEx5_ISO15693.dir/NfcrdlibEx5_ISO15693.c.o' failed
make[2]: *** [NfcrdlibEx5_ISO15693/CMakeFiles/NfcrdlibEx5_ISO15693.dir/NfcrdlibEx5_ISO15693.c.o] Error 1
CMakeFiles/Makefile2:2635: recipe for target 'NfcrdlibEx5_ISO15693/CMakeFiles/NfcrdlibEx5_ISO15693.dir/all' failed
make[1]: *** [NfcrdlibEx5_ISO15693/CMakeFiles/NfcrdlibEx5_ISO15693.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
============================================

any ideas?

sabberlatz, 2018-01-25:
Hi,

I have still difficulties compiling the stuff:

[ 85%] Building C object NfcrdlibEx3_NFCForum/CMakeFiles/NfcrdlibEx3_NFCForum.dir/src/phApp_Init.c.o
[ 85%] Linking C executable NfcrdlibEx3_NFCForum
[ 85%] Built target NfcrdlibEx3_NFCForum
Scanning dependencies of target NfcrdlibEx4_MIFAREClassic
[ 86%] Building C object NfcrdlibEx4_MIFAREClassic/CMakeFiles/NfcrdlibEx4_MIFAREClassic.dir/NfcrdlibEx4_MIFAREClassic.c.o
[ 86%] Building C object NfcrdlibEx4_MIFAREClassic/CMakeFiles/NfcrdlibEx4_MIFAREClassic.dir/src/phApp_Init.c.o
[ 87%] Linking C executable NfcrdlibEx4_MIFAREClassic
[ 87%] Built target NfcrdlibEx4_MIFAREClassic
Scanning dependencies of target NfcrdlibEx5_ISO15693
[ 87%] Building C object NfcrdlibEx5_ISO15693/CMakeFiles/NfcrdlibEx5_ISO15693.dir/NfcrdlibEx5_ISO15693.c.o
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c: In function ‘NfcrdlibEx5_ISO15693’:
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:209:43: error: ‘phacDiscLoop_Sw_DataParams_t’ has no member named ‘sTypeVTargetInfo’
                 phApp_Print_Buff(sDiscLoop.sTypeVTargetInfo.aTypeV[0].aUid, 0x08);
                                           ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:212:47: error: ‘phacDiscLoop_Sw_DataParams_t’ has no member named ‘sTypeVTargetInfo’
                 memcpy(aReceivedUid, sDiscLoop.sTypeVTargetInfo.aTypeV[0].aUid, 0x08);
                                               ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:215:50: error: ‘phacDiscLoop_Sw_DataParams_t’ has no member named ‘sTypeVTargetInfo’
                 if (DisplayCardTypeInfo(sDiscLoop.sTypeVTargetInfo.aTypeV[0].aUid) == PH_ERR_SUCCESS)
                                                  ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:226:62: error: ‘salI15693’ undeclared (first use in this function)
                         status = phalI15693_ReadSingleBlock(&salI15693,
                                                              ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:226:62: note: each undeclared identifier is reported only once for each function it appears in
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:227:29: error: ‘PHAL_I15693_OPTION_OFF’ undeclared (first use in this function)
                             PHAL_I15693_OPTION_OFF,
                             ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:286:26: error: ‘PHPAL_SLI15693_FLAG_NBSLOTS’ undeclared (first use in this function)
                         (PHPAL_SLI15693_FLAG_NBSLOTS | PHPAL_SLI15693_FLAG_DATA_RATE | PHPAL_SLI15693_FLAG_INVENTORY),
                          ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:286:56: error: ‘PHPAL_SLI15693_FLAG_DATA_RATE’ undeclared (first use in this function)
                         (PHPAL_SLI15693_FLAG_NBSLOTS | PHPAL_SLI15693_FLAG_DATA_RATE | PHPAL_SLI15693_FLAG_INVENTORY),
                                                        ^
/home/pi/nxp/NfcrdlibEx5_ISO15693/NfcrdlibEx5_ISO15693.c:286:88: error: ‘PHPAL_SLI15693_FLAG_INVENTORY’ undeclared (first use in this function)
                         (PHPAL_SLI15693_FLAG_NBSLOTS | PHPAL_SLI15693_FLAG_DATA_RATE | PHPAL_SLI15693_FLAG_INVENTORY),
                                                                                        ^
NfcrdlibEx5_ISO15693/CMakeFiles/NfcrdlibEx5_ISO15693.dir/build.make:62: recipe for target 'NfcrdlibEx5_ISO15693/CMakeFiles/NfcrdlibEx5_ISO15693.dir/NfcrdlibEx5_ISO15693.c.o' failed
make[2]: *** [NfcrdlibEx5_ISO15693/CMakeFiles/NfcrdlibEx5_ISO15693.dir/NfcrdlibEx5_ISO15693.c.o] Error 1
CMakeFiles/Makefile2:2635: recipe for target 'NfcrdlibEx5_ISO15693/CMakeFiles/NfcrdlibEx5_ISO15693.dir/all' failed
make[1]: *** [NfcrdlibEx5_ISO15693/CMakeFiles/NfcrdlibEx5_ISO15693.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

dosenfisch, 2018-01-25:
I just found out that "SW3693-ARCHIVED.zip" is the only version of SW3693, that is working as nxp.zip.

sabberlatz, 2018-01-25:
do you have the link to download it for me?
thank you very much!

svvitale, 2018-01-25:
Steps to install (for now):

1. Clone this repo
1. Go to https://www.nxp.com/products/identification-and-security/nfc/nfc-reader-ics/explore-nfc-exclusive-from-element14:PNEV512R?tab=Design_Tools_Tab and download the "NFC Reader Library for Linux".
1. Make sure that the zip you downloaded is named `nxp.zip` and is located in the root of this code repo.  No need to extract it, the install process will do that.
1. Run `sudo python setup.py install` or just `python setup.py install` if you're running in a virtualenv (**recommended**)

michox, 2018-01-25:
i installed the package successfully but if i execute a mifare.select() it just stops there. no error no output. nothing. i would like to reinstall it, but if i delete the folder, python still imports the package. How can i completely remove nxppy from my raspberry?

svvitale, 2018-01-25:
By default, you must run your script as root to access the SPI port.  Try running your script with `sudo`.  To remove the package, use `sudo pip uninstall nxppy` or `pip uninstall nxppy` (if you're in a virtualenv).

michox, 2018-01-25:
Heiner.frueh@ieh.uni-stuttgart.de 
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

Heiner92, 2018-01-26:
I installed nxppy as you said (created an account on nxp.com) but I have the same error as michbox. 

I can import the module nxppy but I cant use the "uid=mifare.select()" method. I have tried running it with root privileges as you suggested. No Difference.
It says "nxppy.InitError: Nxppy: Internal Error from BAL Component"

I have verfied the functionality of the NFC-Module. (I have a sencond one that works on a different Pi but not on mine)

Can you give me any advice? I need this to work rather urgently.

sabberlatz, 2018-01-29:
Sorry, I give up. 

Where I have to copy the nxp.zip? What do you mean with "in the root of this code repo"? I have the path "/home/pi/". There I copied the content of the nxppy-master.zip and so I got the path

/home/pi/nxppy-master/

I copied the nxp.zip you providede in this path and started "sudo python setup.pl install". 

I received the same ugly error messages as follows: 

- - - - - - - - - - - - - - - - - - - - 

**NXP Reader Library found, skipping download
./get_nxpRdLib.sh: line 22: pushd: nxp/build: No such file or directory
-- Configuring done
CMake Error at CMakeLists.txt:290 (add_executable):
  Cannot find source file:

    nxp/addition/kernel-space_bal-module/bal/include/linux/spi/bal_spi.h

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error: CMake can not determine linker language for target: nxppy
CMake Error: Cannot determine link language for target "nxppy".
-- Generating done
-- Build files have been written to: /home/pi/nxppy-master
make: *** No rule to make target 'NxpRdLibLinuxPN512'.  Stop.
./get_nxpRdLib.sh: line 25: popd: directory stack empty
building 'nxppy' extension
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DLINUX -DNATIVE_C_CODE -DNXPBUILD_CUSTOMER_HEADER_INCLUDED -DNXPBUILD__PHHAL_HW_RC523 -I/usr/include/python2.7 -c Mifare.c -o build/temp.linux-armv7l-2.7/Mifare.o -O0 -std=gnu99 -isystemnxp/nxprdlib/NxpRdLib/intfs -isystemnxp/nxprdlib/NxpRdLib/types -isystemnxp/nxprdlib/NxpRdLib -isystemnxp/linux/intfs -isystemnxp/linux/comps/phbalReg/src/Linux -isystemnxp/linux/shared -isystemnxp/examples/NfcrdlibEx4_MIFAREClassic/intfs -isystemnxp/nxprdlib/NxpRdLib/comps/phbalReg/src/Stub -isystemnxp/linux/comps/phPlatform/src/Posix -isystemnxp/linux/comps/phOsal/src/Posix
In file included from Mifare.c:1:0:
Mifare.h:32:24: fatal error: phhwConfig.h: No such file or directory
 #include <phhwConfig.h>
                        ^
compilation terminated.
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1**

- - - - - - - - - - - - - - - - - - - - 

what is wrong here? Why the path "xnp/build" isn´t found but was created successfully? Why phhwConfig.h isn´t found?

have fun!

I<it

dosenfisch, 2018-01-29:
Search for [SW3693-ARCHIVED.zip at NXP](https://www.nxp.com/search?client=nxp_search_all_results&site=nxp_en&proxystylesheet=nxp_search_style_fe&sort=date%3AD%3AL%3Ad1&oe=UTF-8&ie=UTF-8&ud=1&output=xml_no_dtd&exclude_apps=1&callback=ss_show&lang_cd=en&filter=0&getfields=*&baseUrl=%2F%2Fwww.nxp.com%2Fwebapp&SEARCH_OPERATOR=Contains&attempt=-1&rc=1&hl=en&dnavs=&q=SW3693-ARCHIVED.zip). That was the only Version of SW3693 (reader lib) working as NXP.zip.

sabberlatz, 2018-01-30:
Vielen Dank "dosenfisch"!

It´s now working. Here is a summary how it should work:

prepare the system as usual by entering....

sudo apt-get upgrade
sudo apt-get update
sudo apt-get install python-dev

and activate the SPI port by using sudo raspi-config

1. renaming after downloading on a stick the file **SW3693-ARCHIVED.zip** to **nxp.zip**
2. copy this **nxp.zip** file into the already downloaded and extracted "**/home/pi/nxppy-master/**" directory
3. open terminal and change directory to **/home/pi/nxppy-master/**
4. change the attributes of **get_nxpRdLib.sh** to make it executable be entering **chmod 777 get_nxpRdLib.sh**

5. start **sudo ./get_nxpRdLib.sh**
6. start **sudo python setup.py install**
7. start **sudo python3 setup.py install**
8. if python2 or python3 is running, restart it because the nxppy module would not be found otherwise
9. create a simple **test.py** file containing the line
    **import nxppy**
and start it. No error should be shown. 
10. take a beer, put on a smile and give your wife a kiss because she was so patient.

hint: I ran it on a Raspberry 3 under the last version of Raspbian Stretch. My Raspbian Zero, Raspbian 2 and Raspbian will follow.

hint2: I hope other guys spend time by using my description instead of dig in the fog.

have fun!

I<it

Tetrikus, 2018-02-19:
Danke, @sabberlatz
Quite useful step-by-step guide. Unfortunately this isn't working for me. I'm sure I'm doing something wrong.

1. I did clone the repo to /home/pi/nxppy-master
2. I did download the sw3693-Archive and copied it into the ..master root
3. I chmod'ed to 777
4. **sudo ./get_nxpRdLib.sh** tells me the following:

`nxp.zip found, extracting
NXP Reader Library found, skipping download
./get_nxpRdLib.sh: line 22: pushd: nxp/build: No such file or directory
CMake Error: The source directory "/home/pi" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** No rule to make target 'NxpRdLibLinuxPN512'.  Stop.
./get_nxpRdLib.sh: line 25: popd: directory stack empty`

The mentioned CMakeLists.txt is (of course) not in the directory "/home/pi" but in the subfolder "/home/pi/nxppy-master/nxp/"

Any hints? Thanks in advance!

svvitale, 2018-02-20:
@Tetrikus, did you get it going?  I'm working on a longer term fix for this, but it's not quite there yet.

Tetrikus, 2018-02-20:
@svvitale Thanks, yes, I did.. once I managed to do this workaround correctly, everything went smooth.

closed 2018-03-19 by svvitale

svvitale, 2018-03-19:
Thank you for your patience, everyone.  NXP has released a new nxppy-compatible DEB package file.  See the latest README.md for instructions on how to download and install it.  Once it's installed, running ```pip install nxppy``` will get you the latest version of nxppy.  Cheers!

Antjevb, 2018-12-16:
@svvvital Where do I find this README.md for these download and install instructions? The README.md in the nxppy-release-1.5.1 does not include instructions for a nxpp-compatible DEB package file.

svvitale, 2018-12-18:
Hi @Antjevb , you need to move to the 1.6.0 release.  Here are the instructions on installing the NXP pre-requisite: https://github.com/svvitale/nxppy/blob/master/README.md#external-components

Antjevb, 2018-12-19:
Thank you for you answer. I moved to the 1.6.0. release and installed the DEB file succesfully.

commit af268550 (2018-03-18): Update README instructions for new DEB file installation.  Fixes #55.