svvitale/nxppy #57: Reading from AS3955

benkli, 2017-03-21:
Please answer these questions as part of your new issue.

##### Which version of Python are you using?
2.7.9
##### Which OS and version are you running?
Linux raspberrypi 4.4.38-v7+
##### Is SPI enabled?
yes
##### Are you trying to read/write a Mifare Classic card?  (This is currently unsupported)
I'm trying to read an AMS AS3955 NFC interface tag. This tag acts as an NFC type 2 tag (mifare ultralight)

Hey there!
I'm trying to read data from an AMS AS3955 NFC interface tag.
When reading and writing from EEPROM data section 0 -> 127 everthing works fine.
But when I'm trying to read from the AS3955 extendend memory section (0xfc -> 0xff),
I'm getting: 

> nxppy.ReadError: Nxppy: Protocol Error, response violated protocol from Mifare PAL-Component

According to the AS3955 datasheet, I can access this area via normal read and write commands. Writing to this section works, I can read the written data with the MCU attached to the AS3955. But I can not read the data from this section with nxppy.
I've run out of ideas. Thank you in advance for your help!

Yours, Benjamin

benkli, 2017-03-23:
Strange! When reading from memory 0xfc alone, an error does not occure. But the returned data is always 0.

svvitale, 2017-08-10:
Was the `MFC_BLOCK_DATA_SIZE` the only change you made?  Any other side effects that you can tell?  If not, I'd like to merge your changes into master.  Mind filing a pull request?

benkli, 2017-08-10:
> Antworten an: svvitale/nxppy <reply@reply.github.com>

Hi Scott!
I copied the mifare_read(...) function and made a second one named mifare_read16(...). This one had the changed MFC_BLOCK_DATA_SIZE=16 in it. This solution enabled me to interface the AS3955’s buffer. I Couldn’t recognize any side effects. 

I’m an electrical engineer and not very familiar with the git terminology, so I’m not quite sure what a pull request is. J

 

With best regards

Benjamin

svvitale, 2017-08-10:
Ah, no problem.  I'll give it a try using your description and we can go from there.  I'll ping you again when I need a guinea pig :).  I'll reopen this issue since it is something that needs to be fixed.

commit c9424e06 (2018-03-14): Fixes #68, fixes #57 - consistently read blocks in 16 byte increments, but still only return one page of data (4 bytes) from the read_block() and pwd_auth() methods.
closed 2018-03-19 by svvitale