svvitale/nxppy #60: Send a file from Raspberry PI to Android

P1K3N, 2017-05-04:
Please answer these questions as part of your new issue.

##### Which version of Python are you using?

2.7.9

##### Which OS and version are you running?
Raspbian

##### Is SPI enabled?
Yes

##### Are you trying to read/write a Mifare Classic card?  (This is currently unsupported)
Dont know 

###### Hello guys
I'm desperate to understand this lib, but I really can't get it. 
I have this explore-nfc and a RPI3. 
When a Tag from an Android phone is detected i want to send a file from the RPI3 to it and then I want that RPI3  waits for a TAG from this Android Phone which is a JSON file. Is it possible? HOw can I do it?
I've been for days try to understand how to incorporate this lib with ndeflib but without success.

Guys please help me!! 
 I Really need help...

sherylll, 2017-05-09:
I think this is not possible with this lib (nxppy) though. What you need should be P2P mode, check out the examples that come with the original NXP reader library.

It would be cool to wrap the functions related with P2P mode from the library, but I don't know how difficult it is to do that... Wish someone could help too.

Otherwise there is still nfcpy, which apparently doesn't support explore-nfc :(

Another workaround, since you have Pi3, is to set up your pi as an access point, use NFC to start a session between your phone and your pi and transmit the file over wifi (don't know if it makes sense for your project)

Good luck, and I'd love to know if you have any new ideas about the Android-to-RPI part :)

svvitale, 2017-05-09:
Hi @P1K3N,

As @sherylll mentioned, nxppy has only very limited functionality focused on reading and writing tags.  I believe for what you're describing, you can take a look at the Card Emulation example in the [NXP Start-up Guide](http://www.nxp.com/documents/application_note/AN11480.pdf).  This will allow you to use the EXPLORE-NFC as an emulated "card" and then read from that card using your smart phone.  Your card can contain NDEF style data.  As long as your payload fits within the card storage, I think this will work for your purpose.

I hope to add more functionality to nxppy in the future including these emulation use cases.  Unfortunately I don't have much time to work on it right now.  If you're able to get it working, pull requests are always welcome.

I'll leave this issue open for now, since this is a feature I'd like to incorporate.  Thanks.