svvitale/nxppy #61: Reading Data from Tag ISO14443 -4B

Salabansi, 2017-05-18:
Please answer these questions as part of your new issue.

##### Which version of Python are you using?
Python3
##### Which OS and version are you running?
Linux
##### Is SPI enabled?
Yes
##### Are you trying to read/write a Mifare Classic card?  (This is currently unsupported)
No

Salabansi, 2017-05-18:
Hey Guys,

is there any python wrapper which I cann use to read Tags with ISO14443 4B standard?

Nxppy https://github.com/svvitale/nxppy/blob/master/README.md  is only using ISO144443 3A/4A standards. Or is there a way to reuse this software??

Thanks in advance.

Cheers

svvitale, 2017-07-27:
Hi @Salabansi, from what I can tell, this can be implemented within nxppy.  Unfortunately I don't have any 4B tags to test with.  Would it be possible for you to mail me a few samples?

manojrohit, 2017-07-28:
Hello @svvitale. The NXP Reader Library has no APIs to read the NDEF message to read type 4B tags. However the UID of the type 4b tags can be read using few APIs but not NDEF.

svvitale, 2017-07-28:
I admit I haven't looked that closely, but given that the PN512 supports 14443/4B tags, I assumed there'd be a way to read the data pages in addition to the UID.  Currently, nxppy uses this same method on 4A tags and then parses the data into NDEF messages.

svvitale, 2018-03-13:
Created a separate issue to track enhancements to this functionality (#71).

closed 2018-03-13 by svvitale