svvitale/nxppy #62: Error while installing nxppy

Salabansi, 2017-05-22:
Please answer these questions as part of your new issue.

##### Which version of Python are you using?
3
##### Which OS and version are you running?
Linux 8 (jessie)
##### Is SPI enabled?
yes
##### Are you trying to read/write a Mifare Classic card?  (This is currently unsupported)
yes

svvitale, 2017-05-22:
What's the error you're seeing?

Salabansi, 2017-05-22:
Fatal Error while i´m running sudo pip install nxppy:

Can´t find Python.h in directory.

What should i do? I downloaded the content and tried to install it by cmake&make, but there is also an error.
Any solvings??

Thx
Bassem

svvitale, 2017-05-22:
You're missing the python-dev package for your version of python. Please `apt-get install python3-dev` or `python2.7-dev` as appropriate.

Salabansi, 2017-05-23:
Thank you!

I already got python3. After I installed python2.7-dev it´s working.
Is nxppy ready to read ISO14443 Type B Tags??

Greez

csaeum, 2018-03-12:
I get the following error when running Raspy3 with Stretch while executing pip install nxppy

The .deb I could install and there the NFC test script works but the script of small.li breaks off with missing nxppy.

===
Collecting nxppy
  Using cached nxppy-1.5.1.tar.gz
Building wheels for collected packages: nxppy
  Running setup.py bdist_wheel for nxppy: started
  Running setup.py bdist_wheel for nxppy: finished with status 'error'
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-o3n5da8w/nxppy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpmapa0s65pip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  running build_ext
  compiling NxpRdLib
  [[0;34mDownloading NXP Reader Library[0m]
  --2018-03-12 16:38:42--  https://nxp1.sharepoint.com/sites/6_1/S&CExternal/_layouts/15/guestaccess.aspx?docid=0afe70cb96bba406d9c3c27534d03087d&authkey=AU04PywGBtgqJtk4Fp0GT0M
  Auflösen des Hostnamens »nxp1.sharepoint.com (nxp1.sharepoint.com)« … 2a01:111:f402:9412::28, 13.107.6.151
  Verbindungsaufbau zu nxp1.sharepoint.com (nxp1.sharepoint.com)|2a01:111:f402:9412::28|:443 … verbunden.
  HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
  Länge: 22232 (22K) [text/html]
  Wird in »»nxp.zip«« gespeichert.
  
       0K .......... .......... .                               100% 8,65M=0,002s
  
  2018-03-12 16:38:43 (8,65 MB/s) - »»nxp.zip«« gespeichert [22232/22232]
  
  [nxp.zip]
    End-of-central-directory signature not found.  Either this file is not
    a zipfile, or it constitutes one disk of a multi-part archive.  In the
    latter case the central directory and zipfile comment will be found on
    the last disk(s) of this archive.
  unzip:  cannot find zipfile directory in one of nxp.zip or
          nxp.zip.zip, and cannot find nxp.zip.ZIP, period.
  ./get_nxpRdLib.sh: Zeile 17: pushd: nxp/build: Datei oder Verzeichnis nicht gefunden
  CMake Error: The source directory "/tmp/pip-build-o3n5da8w" does not appear to contain CMakeLists.txt.
  Specify --help for usage, or press the help button on the CMake GUI.
  make: *** Keine Regel, um „NxpRdLibLinuxPN512“ zu erstellen.  Schluss.
  ./get_nxpRdLib.sh: Zeile 20: popd: Der Verzeichnisstapel ist leer.
  building 'nxppy' extension
  creating build
  creating build/temp.linux-armv7l-3.5
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-RUbMX3/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DLINUX -DNATIVE_C_CODE -DNXPBUILD_CUSTOMER_HEADER_INCLUDED -DNXPBUILD__PHHAL_HW_RC523 -I/usr/include/python3.5m -c Mifare.c -o build/temp.linux-armv7l-3.5/Mifare.o -O0 -std=gnu99 -isystemnxp/nxprdlib/NxpRdLib/intfs -isystemnxp/nxprdlib/NxpRdLib/types -isystemnxp/nxprdlib/NxpRdLib -isystemnxp/linux/intfs -isystemnxp/linux/comps/phbalReg/src/Linux -isystemnxp/linux/shared -isystemnxp/examples/NfcrdlibEx4_MIFAREClassic/intfs -isystemnxp/nxprdlib/NxpRdLib/comps/phbalReg/src/Stub -isystemnxp/linux/comps/phPlatform/src/Posix -isystemnxp/linux/comps/phOsal/src/Posix
  In file included from Mifare.c:1:0:
  Mifare.h:32:24: fatal error: phhwConfig.h: Datei oder Verzeichnis nicht gefunden
   #include <phhwConfig.h>
                          ^
  compilation terminated.
  error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
  
  ----------------------------------------
  Running setup.py clean for nxppy
Failed to build nxppy
Installing collected packages: nxppy
  Running setup.py install for nxppy: started
    Running setup.py install for nxppy: finished with status 'error'
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-o3n5da8w/nxppy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-lin4as6i-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_ext
    compiling NxpRdLib
    [[0;34mDownloading NXP Reader Library[0m]
    --2018-03-12 16:38:47--  https://nxp1.sharepoint.com/sites/6_1/S&CExternal/_layouts/15/guestaccess.aspx?docid=0afe70cb96bba406d9c3c27534d03087d&authkey=AU04PywGBtgqJtk4Fp0GT0M
    Auflösen des Hostnamens »nxp1.sharepoint.com (nxp1.sharepoint.com)« … 2a01:111:f402:9412::28, 13.107.6.151
    Verbindungsaufbau zu nxp1.sharepoint.com (nxp1.sharepoint.com)|2a01:111:f402:9412::28|:443 … verbunden.
    HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
    Länge: 22232 (22K) [text/html]
    Wird in »»nxp.zip«« gespeichert.
    
         0K .......... .......... .                               100% 5,02M=0,004s
[... 8 lines cut ...]
    unzip:  cannot find zipfile directory in one of nxp.zip or
            nxp.zip.zip, and cannot find nxp.zip.ZIP, period.
    ./get_nxpRdLib.sh: Zeile 17: pushd: nxp/build: Datei oder Verzeichnis nicht gefunden
    CMake Error: The source directory "/tmp/pip-build-o3n5da8w" does not appear to contain CMakeLists.txt.
    Specify --help for usage, or press the help button on the CMake GUI.
    make: *** Keine Regel, um „NxpRdLibLinuxPN512“ zu erstellen.  Schluss.
    ./get_nxpRdLib.sh: Zeile 20: popd: Der Verzeichnisstapel ist leer.
[... cut ...]
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    
    ----------------------------------------

  Failed building wheel for nxppy
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-o3n5da8w/nxppy/setup.py';f=geta
ttr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exe|
c'))" install --record /tmp/pip-lin4as6i-record/install-record.txt --single-version-externally-managed --compile" fai
led with error code 1 in /tmp/pip-build-o3n5da8w/nxppy/

svvitale, 2018-03-13:
@csaeum see #55.  We're close to releasing a better solution for this issue with the help of NXP.

svvitale, 2018-03-13:
@Salabansi nxppy hasn't been written specifically to work with Type B tags.  From what I can find in the NXP forums, the underlying PN512 does support Type B, but it may require some additional work in the C layer.  I've added this as a separate feature request (#71) and I'll close this issue since your installation issues were resolved.

closed 2018-03-13 by svvitale