svvitale/nxppy #68: mifare.read() stop executing of program

sabberlatz, 2018-01-30:
Please answer these questions as part of your new issue.

##### Which version of Python are you using?
3 and 2.7

##### Which OS and version are you running?
Raspberry Stretch
##### Is SPI enabled?
yes
##### Are you trying to read/write a Mifare Classic card?  (This is currently unsupported)
I have the issue, that mifare.read() stop the program without any error message. The code after the command isn´t processed afterwards.
I used the NFC sample card which is shipped with the card reader. 

I have two other "white" cards, which are shipped with the other RFID-RC522 reader. 

mifare.select() give me a unique number for each card, but it seems, that writing and re-reading of the card isn´t possible.

So, is the issue caused by using the wrong card type?

have fun!

I<it

sabberlatz, 2018-01-31:
Yes, I found out, it´s card related. With my new Mifare ultralight chips it´s working. I can write and read 4 byte segments.  

have fun!
I<it

svvitale, 2018-03-13:
Sounds like this may have triggered a segmentation fault in the underlying C code.  Would you be willing to mail me one of these "white" cards so I can reproduce and fix the bug?  I'm based in Denver, Colorado, USA.  Send me an email (svvitale@gmail.com) and we can coordinate.

sabberlatz, 2018-03-13:
Hello Scott,

 

I assume it´s easier for you ordering such card from ebay. Here is one of the offer where you can see the card. You should be able to buy such card separately. My card is lost anywhere in my house 😊

 

https://www.ebay.com/itm/PN532-NFC-Module-RFID-Reader-KIT-for-Arduino-Raspberry-Pi-3-3-5-5V/322672095344?_trkparms=aid%3D555017%26algo%3DPL.CASSINI%26ao%3D1%26asc%3D20160706104836%26meid%3D73827c40a3f64b07b37028ddf0f282e1%26pid%3D100642%26rk%3D1%26rkt%3D1%26%26itm%3D322672095344 <https://www.ebay.com/itm/PN532-NFC-Module-RFID-Reader-KIT-for-Arduino-Raspberry-Pi-3-3-5-5V/322672095344?_trkparms=aid%3D555017%26algo%3DPL.CASSINI%26ao%3D1%26asc%3D20160706104836%26meid%3D73827c40a3f64b07b37028ddf0f282e1%26pid%3D100642%26rk%3D1%26rkt%3D1%26%26itm%3D322672095344&_trksid=p2045573.c100642.m3226> &_trksid=p2045573.c100642.m3226

 

 

Mit freundlichen Grüßen!

 

Stephan Kittel

svvitale, 2018-03-13:
I'm guessing this is a Mifare Ultralight and there's a known issue with the `read()` method and these types of cards.  They don't properly indicate their size in the card memory, so we end up reading off the end.  Thanks for jogging my memory, I'll resolve this soon.

commit c9424e06 (2018-03-14): Fixes #68, fixes #57 - consistently read blocks in 16 byte increments, but still only return one page of data (4 bytes) from the read_block() and pwd_auth() methods.
closed 2018-03-19 by svvitale