svvitale/nxppy #73: Mifare Init Error

Keriz, 2018-04-14:
Please answer these questions as part of your new issue.

##### Which version of Python are you using?
Python 2.7
##### Which OS and version are you running?
Armbian 5.38 Ubuntu 16.04 LTS 3.4.112-sun8i
##### Is SPI enabled?
Yes
##### Are you trying to read/write a Mifare Classic card?  (This is currently unsupported)
No

Hello,
I followed all the steps on the readme and the installation process was rather smooth (good job on that)!

So I tried the polling example but it seems that I cannot get it to run.

I removed everything and just put the `import nxppy`.

It works.

Added the mifare = nxppy.Mifare() line, and here I get an error: 
`nxppy.InitError: Nxppy:Unknown Error from Undefined Component`

I've been reading all the issues and no one seems to solve my problem. I hope there is a solution for this!

svvitale, 2018-04-14:
Hi @Keriz,

I haven't seen this one before.  Are you running the polling example as root?

Keriz, 2018-04-14:
Thanks for the quick answer. Yes I am running the polling example as root.

svvitale, 2018-04-14:
Is this on a Raspberry Pi or an Orange Pi?  I don't see any Armbian builds for Raspberry Pi.  Forgive me if that's a stupid question - I'm not familiar with this distro.

Keriz, 2018-04-14:
This is on an Orange Pi, sorry for not precising it in my post, as it might be the cause of the problem (but why?)

Maybe it is a hardware specific issue?

thibww67, 2020-08-24:
It's a long time ago but did you find a solution.. I am desperated !

Thunder8Storm, 2020-08-30:
I have the same error on a new installed raspberry 3b. Any ideas to fix this?
[image]

Keriz, 2020-08-30:
@thibww67 @Thunder8Storm Sorry for the late reply. It's a long time ago and I don't remember finding a solution to this.
Maybe try changing of distribution, or dive into the nxxpy code if you feel capable.

thibww67, 2020-08-30:
> I have the same error on a new installed raspberry 3b. Any ideas to fix this?
> [image]

For my part, I had this error when I worked on a RPi used for an other application. But when I tried to work on a totally new version, it works. Try to just installed the libraries that you need for nxppy.

Thunder8Storm, 2020-08-31:
@thibww67 I am using a fresh Raspberry Pi OS, downloaded yesterday with their installer / imager for Windows. I tried reinstall two times once with gui once without. Both not working. Which OS did you use?

thibww67, 2020-08-31:
@Thunder8Storm I use Raspbian Lite for my application, but I don't know the real problem behind this error so it's complicated..

Thunder8Storm, 2020-08-31:
@thibww67 I got it work today with an image of Raspian Jessie. Thank you for your help!

incubo4u, 2020-09-29:
The error exists in my case on almost every distribution for rpi 4 4ram

AlexanderLaix, 2020-12-13:
> The error exists in my case on almost every distribution for rpi 4 4ram
... Almost? Which is working?
I have the same issue I guess https://github.com/svvitale/nxppy/issues/80

mac2612, 2022-06-06:
Hi, on my Pi 2 this was a GPIO conflict - the NXP driver tries to grab GPIO7, which is already taken for cs1 (chip select 1) in the SPI driver.

After switching to single chip select setup, nxppy works without issue.

In /boot/config.txt, right after the line 'dtparam=spi=on', add this line:
dtoverlay=spi0-1cs

Then, reboot the pi. I have not tried this on a pi4, but I suspect it's same/similar issue. If you're trying to debug, you can use strace with the nxppy sample code (I used the polling code) to see what it's trying to request, and /sys/kernel/debug/gpio has a list of which gpio lines are in use and who is using them.