svvitale/nxppy #76: bug in nxp lib?

iceman1001, 2018-10-20:
Please answer these questions as part of your new issue.

##### Which version of Python are you using?
v2.7
##### Which OS and version are you running?
PI 
##### Is SPI enabled?
yes
##### Are you trying to read/write a Mifare Classic card?  (This is currently unsupported)
no

There seems to be a bug in the NxpLib.  
--

Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x76aa47e8 in phbalReg_Stub_Exchange ()
   from /home/pi/.local/lib/python2.7/site-packages/nxppy.so
(gdb) bt
#0  0x76aa47e8 in phbalReg_Stub_Exchange ()
   from /home/pi/.local/lib/python2.7/site-packages/nxppy.so
#1  0x76a9127c in phhalHw_Rc523_ReadRegister ()
   from /home/pi/.local/lib/python2.7/site-packages/nxppy.so
#2  0x76a8e2e0 in phhalHw_Rc523_FieldOff ()
   from /home/pi/.local/lib/python2.7/site-packages/nxppy.so
#3  0x76a7fad0 in Mifare_select (self=0x3d7368) at Mifare.c:289
#4  0x0008be70 in PyEval_EvalFrameEx ()
#5  0x00084494 in PyEval_EvalCodeEx ()
#6  0x0008c8dc in PyEval_EvalFrameEx ()
#7  0x0008c7d0 in PyEval_EvalFrameEx ()
#8  0x0008c7d0 in PyEval_EvalFrameEx ()
#9  0x00084494 in PyEval_EvalCodeEx ()
#10 0x000841f8 in PyEval_EvalCode ()
#11 0x000bd8b8 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

svvitale, 2018-10-21:
@michaelneurohr, can you verify?  I agree with @iceman1001 that this line looks fishy.

iceman1001, 2018-10-22:
The bug is triggered in my stress tests,  when sending a select/write sequence in a loop with a sleep 1sec each iteration. It takes around 240-320 times before triggered.  If the loop has a sleep for two seconds or more the loop can run to 3000 times at least.  
Could it be spi transfer being called too fast?