svvitale/nxppy #78: Import nxppy error: Import Error: nxppy.so: undefined symbol: Py_InitModule4

Antjevb, 2018-12-19:
##### Which version of Python are you using?
Python 3.5.3

##### Which OS and version are you running?
Linux / Raspberry Pi 3B+ / Rasbian

##### Is SPI enabled?
Yes

##### Are you trying to read/write a Mifare Classic card?  (This is currently unsupported)
No

When I try to import the nxppy module, the Python Shell responds with this Import Error:
"Import Error: /[path]/nxppy.so: undefined symbol: Py_InitModule4".
I can not open this nxppy.so file.

I tried to find more information about [this import error](https://stackoverflow.com/questions/12589348/importerror-undefined-symbol-py-initmodule4-for-swig), but in Python 3.0 this modsupport.h file does not mention Py_InitModule4.

Any help to make the nxppy module work would be greatly appreciated.

svvitale, 2018-12-19:
Hi @Antjevb,

Can you verify that you built nxppy.so and are running it with the exact same version of Python?  Sounds like it may have been built for 2.x and run on 3.x or vice versa.  You may have to use `pip3 install nxppy` to build it using Python 3.

Antjevb, 2018-12-20:
Hey @svvitale ,

`pip3 install nxppy` results into:
```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: nxppy in /[path]/python3.5/dist-packages (1.6.0)
```

Antjevb, 2018-12-29:
Update:
After a clean installation of Raspbian I followed the instructions as you suggested here at Github.
It works like a charm!
I don't know what went wrong before, but I must have messed up something to get these previous results. It might have something to do with installing both Python 2.7 and Python 3.0.

svvitale, 2019-01-02:
Glad to hear you got it resolved.

closed 2019-01-02 by svvitale