svvitale/nxppy #79: Write Ndef on a tag

thibww67, 2020-08-18:
Please answer these questions as part of your new issue.

##### Which version of Python are you using?
Python 3.7.3

##### Which OS and version are you running?
Raspbian Lite on RPi

##### Is SPI enabled?
Yes

##### Are you trying to read/write a Mifare Classic card?  (This is currently unsupported)
No

I'm trying to write a text as a Ndef message with my Explore-NFC. I see that we can read easily Ndef message but how can I write it properly ?

Thank you !

svvitale, 2020-08-19:
Hi @thibww67,

Yes, `write_block()` is the correct way to go about this.  First, if the tag is not already "formatted" as NDEF, you'll need to write the "capability container":

```
mifare.write_block(3, "\xe1\x10\x12\x00")
```

Once that's done, you can write the contents of your full NDEF message from block 4 on.  If you're using `ndeflib`, that should look something like this:

```
import ndef

BLOCK_SIZE = 4
STARTING_BLOCK = 4

record = ndef.TextRecord("testing123")
bytes_to_write = b''.join((ndef.message_encoder([record])))
blocks_to_write = len(bytes_to_write) / BLOCK_SIZE

for block in range(STARTING_BLOCK, STARTING_BLOCK + blocks_to_write):
  mifare.write_block(block, bytes_to_write[block * BLOCK_SIZE:BLOCK_SIZE]
```

I wrote that mostly from memory, so let me know if you run into any issues.

thibww67, 2020-08-20:
Thank you for your quick answer !

I will try it soon and let you know if it works.
Now I will try to see if it is possible to use 2 Explore NFC on a single RPi, but I think it is not because of the software.

svvitale, 2020-08-20:
I don't think you can easily multiplex the SPI connections.  You could
potentially add a switch board to toggle between two Explore boards, but I
don't believe there's a way to use them simultaneously without a ton of
work.  ACR122U's are also compatible with the Pi if a USB solution is
possible in your situation.

-Scott

thibww67, 2020-08-21:
Yes, that what I thought, for my application I think I will use the Adafruit NFC Breakout Board, and with its library for python, I can use multiple modules on the SPI.
But thank you for your help !

thibww67, 2020-08-24:
I reflashed it on a new raspbian Lite and it works perfectly !

closed 2020-08-24 by thibww67

thibww67, 2020-08-25:
Concerning the code you dropped, I had to correct some things, for example block_write can't write a message different of 4 bytes, so now I can right anything in the memory but it is not recognize as an Ndef, however the message is correctly encode. I keep working on it !